bmd-studio/stm32-for-vscode #9: Add folder filter for makefile generation

Bonnee, 2019-11-18:
I have a stm32 project that links to a submodule containing cpp code and when I run a build, stm32-for-vscode includes those cpp files in the makefile, breaking the build process.
I'm suggesting to add a filter configuration to select excluded directories inside the project

jortbmd, 2019-11-19:
Hi thanks for opening an issue. I will take this into account when doing an upgrade as I think it would be a helpful feature. It will probably result in an .stmignore file using glob patterns for file or folder exclusion.

jortbmd, 2020-05-06:
Added a new .stmignore functionality which allows you to use glob patterns to ignore files. It is in the current release.

closed 2020-05-06 by jortbmd