bmd-studio/stm32-for-vscode #10: Add support for simultaneous devices

Bonnee, 2019-11-22:
The extension doesn't handle when there are multiple devices connected to the same computer.

jortbmd, 2019-11-23:
Hi thanks for opening an issue. Multi device support is not supported directly on the roadmap, as a lot of this is automatically done through openocd. However a quick search would indicate it would be possible with some scripting: https://www.bergzand.net/programming-with-multiple-st-link-programmers.html. 

I can imagine it will be valuable to handle these kind of cases especially when doing communication between targets. Would you need debugging for both targets or is it just the flashing of the targets? If flashing is the issue then perhaps using an external tool or the drag and drop functionality of STM for bin and elf files might be an option.

Bonnee, 2019-11-25:
I'm currently flashing/debugging on multiple devices. As of now, I'm handling this cases by temporarily unplugging one device while flashing/debugging the other. It's not practical or pretty but it works.
Looking at the link you provided it seems that this feature is achievable, but I don't know if it's worth the effort...

jortbmd, 2019-11-29:
This sounds like a minor issue. I agree that the effort pay off might not be worth if for now. I might revisit this in the future however for now I will close this issue.

closed 2019-11-29 by jortbmd