bmd-studio/stm32-for-vscode #11: Allow customisation of openocd commands

jncronin, 2019-11-26:
Can the generated Makefile be changed to use custom OpenOCD commands for the flash and erase targets?  In particular we use a programmer other than ST-Link, thus the hardcoded "-f interface/stlink-v2-1.cfg" parameter in CreateMakefile.js is not helpful.  This could be implemented as a vscode config option that defaults to the current setting?

jortbmd, 2019-11-29:
Hi,
Yes this is possible. I will shortlist this for a next release.

jncronin, 2019-11-29:
Thank you very much.

chasd00, 2020-02-19:
I use a SEGGER j-link edu and would like this feature too

jortbmd, 2020-07-06:
Hi everybody,

Just as a quick update. I have been doing some extensive refactoring to allow for easier community input and feature implementation. 

What I completely forgot to post earlier is that you can easily add a configuration to the launch.json file, which would allow you to debug your application and upload. 

For segger j-link edu this would be (do not forget to replace the xxxYOUR_PROJECT_NAMExx and xxxYOUR_STM_TARGETxxx with your own projects name and stm target chip respectively):
 ```
    {
      "showDevDebugOutput": true,
      "cwd": "${workspaceRoot}",
      "executable": "./build/xxxYOUR_PROJECT_NAMExxx.elf",
      "name": "Debug STM32 jlink",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "preLaunchTask": "Build STM",
      "device": "jlink",
      "configFiles": [
        "interface/jlink.cfg",
        "target/xxxYOUR_STM_TARGETxxx.cfg"
      ]
    }
```

For programming you can use the following terminal command:
```openocd -f interface/jlink.cfg  -f target/xxxYOUR_STM_TARGETxxx.cfg -c "program ./build/xxxYOUR_STM_TARGETxxx.elf verify reset exit"```

jortbmd, 2020-09-07:
As a quick update, you can now chose a different open ocd interface in the settings (I would recommend using the workspace settings and not your global settings). Be sure to use the put interface/ before your actual interface for now. So when using jlink you just put: interface/jlink.cfg into your workspace settings and it will use this each time for flashing and debugging.

As this feature is finally here I am closing this issue.

closed 2020-09-07 by jortbmd