bmd-studio/stm32-for-vscode #13: recipe for target 'build' failed

QAQddbest, 2019-12-21:
First of all, thanks for your works! It's really helpful!
But today I make a new project as usual, the vscode show me this:

```
➜  aaa make -f STM32Make.make
mkdir build
process_begin: CreateProcess(NULL, mkdir build, ...) failed.
make (e=2): €
STM32Make.make:197: recipe for target 'build' failed
make: *** [build] Error 2
```

QAQddbest, 2019-12-21:
Sorry for boring you! I find the problem. It is because of the win10.

closed 2019-12-21 by QAQddbest

ZhongGuanbin, 2024-01-10:
I encountered the same issue.There is no mkdir.exe in Windows. The solution to this problem is also quite straightforward: by installing Git, we have also installed Git Bash, which provides a plethora of Unix commands (such as rm, cp, mv). Simply executing make within Git Bash will circumvent such issues.
Easy to fix it:
``` setting.json
// setting.json
    "terminal.integrated.defaultProfile.windows": "Git Bash"
```