bmd-studio/stm32-for-vscode #15: c_cpp_properties.json overwritten evey time build task is called

prying, 2020-02-13:
In a project ive had to add custom locations to the "includePath" but each time i call the build task i lose them. Also on a side note when shelljs cant figure out the compiler for "compilerPath" it returns "null" which vscode complains about (tired to fix but i have no clue how to build this project, never touched js before).

kasloV, 2020-03-08:
Same here. This is so annoying when you have configured your intelisense but after building it resets.

jortbmd, 2020-03-09:
Hi,
I will look into this. Also the extension should mention it did not find a compiler, otherwise it will not be able to build the files. I added a setup flow for this, but would be interesting to know if you can still build.

kasloV, 2020-03-09:
Hi,
Thanks for replay. I have installed all dependencies mentioned in readme. I'm able to build and flash without errors. The issue I noticed is that files in .vs folder are being replaced with default ones every time I run build task or "build stm32 project" command. 
I need to add gnu toolchain libraries path and gnu-arm-none-eabi-gcc compiler path in c_cpp_properties.json to get intelisense working without "include" errors. Every time I build those paths are being removed.

jortbmd, 2020-05-06:
This issue has been solved in the new version so I'am closing it.

closed 2020-05-06 by jortbmd