bmd-studio/stm32-for-vscode #17: Default build configuration keeps popping up

knandersen, 2020-03-15:
Disclaimer: I'm very new to using VSCode for embedded work, and still a relative novice when it comes to VSCode.

I want to use a different configuration than the default build configuration which uses openocd. But every time I add my own configuration, the default one keeps getting restored and added. And problem is that for some reason, the `> Build and flash to an STM32 platform`-command insists on using the default configuration rather than my custom one.

Have I missed something? Is there a way to just have my configuration in there, or defining which configuration the `> Build and flash to an STM32 platform`-command should use? Thanks!

jortbmd, 2020-03-16:
Hi.

You can use both if you look at the debugger you would see two configurations. One is yours the other is from STM32 For VSCode. You can use f5 to start a debug session. Do keep in mind that it does not build before debugging in your configuration. You might want to add the

closed 2020-03-16 by jortbmd

knandersen, 2020-12-15:
Sorry I forgot to reply to this, @jortbmd ! Thanks for providing assistance on this, I didn't really have a good grip on VSCode or extension settings at the time, so this helped a lot. Also thanks for providing a great tool!