bmd-studio/stm32-for-vscode #19: Makefile generation adds ./ as the arm gcc path

prying, 2020-04-10:
./ is added to the build command i.e `./arm-none-eabi-gcc -c -mcpu=cortex-m0 -mthumb... ect` causing `'.' is not recognized as an internal or external command` error.
For some reason this line here:
https://github.com/bmd-studio/stm32-for-vscode/blob/6ca3bbeba5dd5cc6753e2021ab7e47df19cd6d76/src/CreateMakefile.js#L115
is generating `GCC_PATH=.`

jortbmd, 2020-04-10:
That is weird. It shouldn't do that. However if the path is configured as: "./" it will give this error. Could you perhaps check your settings.json? It should be fixed if you remove the ./ if not let me know.

jortbmd, 2020-05-06:
Closing this due to inactivity. Please notify me and create a new issue should this error persist.

closed 2020-05-06 by jortbmd