bmd-studio/stm32-for-vscode #21: Customizing Makefile variables

dwalkes, 2020-05-10:
Hi,
Thanks for sharing this great project!

I'm wondering if there is a recommended or supported way to customize make variable values.  For instance, I'd like to change the [OPT](https://github.com/bmd-studio/stm32-for-vscode/blob/4ba299aa15e9e547671385e6736b1ef135570c60/src/CreateMakefile.js#L85) value for the [DEBUG](https://github.com/bmd-studio/stm32-for-vscode/blob/4ba299aa15e9e547671385e6736b1ef135570c60/src/CreateMakefile.js#L171) configuration to use `-O0` so source code stepping works as expected.  I tried specifying 
```
			"args": [
				"OPT=-O0"
			],
```
in my `Build STM` and `Build Clean STM` tasks within tasks.json, but it doesn't appear this does anything, presumably because the [Build STM task does not support make arguments](https://github.com/bmd-studio/stm32-for-vscode/blob/db390d82cde19ec7393a3ed1af881cbfa3d49cf0/src/BuildTask.js#L57).  I think since make variables are set with `=` (and make isn't invoked with -e) this means they [can't be set with environment variables either](https://www.gnu.org/software/make/manual/html_node/Environment.html).

If there's already a way to do this that I'm not seeing I'd appreciate a pointer.

Thanks again for sharing and for any suggestions you can provide.

jortbmd, 2020-05-11:
Hi,

Thanks! For now there is no way to pass arguments along. However it is on the roadmap. I will probably implement this very soon. For now the standard compilation optimization is Og (debug), which is the default for debugging so this should also allow you to step through the source code as expected. 

If you still want to change the optimization level. The build process outputs a makefile: "STM32Make.make", which you could use with make with the optimization set to O0 using: 
``make -f STM32Make.make -e OPT=-O0 ``

Hope this helps. I will leave this issue open until I have implemented the functionality, so you will be notified once it is implemented.

jortbmd, 2021-04-13:
Optimization selection and build flags will be included in the upcoming 3.0.0 release. For this reason I am closing this issue.

closed 2021-04-13 by jortbmd

bhatasaka, 2021-06-20:
Hi, is there somewhere I can learn about how to change the optimization selection and build flags in version 3? I have been doing the method mentioned above to build with -Os and c++ 17 and a couple other flags. However, as the makefile is easily overwritten, I think it would be better to have this in a more permanent configuration file, if that exists.

jortbmd, 2021-06-20:
Hi I do not have a specific knowledge base at this moment. Might be something nice to add to the planning. For now ask away in the issue section.
In version 3.+ there is a new way to configure your project. An STM32-for-VSCode.config.yaml file is created on first compilation. The optimization can be changed in there and compiler flags can be added over there. I can recommend looking at the file as it tries to be self explanatory in terms of what you can configure.

If anything is missing that you would like to add in the configuration file let me know.

bhatasaka, 2021-06-21:
> Hi I do not have a specific knowledge base at this moment. Might be something nice to add to the planning. For now ask away in the issue section.
> In version 3.+ there is a new way to configure your [...]

This is exactly what I was looking for, I was able to change the optimization flag no problem, thanks for the direction!. However, I am having some trouble adding cxxFlags. After adding just one flag for testing:
```c++
# Compiler flags
cFlags: []
cxxFlags:
  - -std=c++11
```

And rebuilding, it doesn't seem to add it to the STM32Make.make or to use the flags in the actual building. Here is the g++ command run:
``` powershell
"C:/PROGRAM FILES (X86)/GNU ARM EMBEDDED TOOLCHAIN/10 2020-Q4-MAJOR/BIN/arm-none-eabi-g++" -c  -feliminate-unused-debug-types -mcpu=cortex-m0plus -mthumb   -DSTM32L051xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32L0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32L0xx_HAL_Driver/Inc 
-IDrivers/STM32L0xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Core/Src/main.cpp -o build/main.o
```

I just tried this out on a fresh STM32L051K8 project from STM32CubeMX, configured in the way the extension specifies. The only modification I made was changing main.c to main.cpp. I tried deleting the STM32Make.make file, then rebuilding. I also tried a different yaml syntax from an online validator. The flags just don't seem to get added to the Makefile - do I have the syntax incorrect, or is there some other core misunderstanding that I have? Manually adding the flags to the generated makefile produces the commands I'm expecting.

According to vscode, the extension is version 3.1.1

Let me know if you need any more info. Thanks! And thank you for putting this extension together as well as supporting it - it has been a pleasure to work with, especially compared to modifying all of the vscode files and Makefiles by hand.

jortbmd, 2021-06-22:
Hi! The info is more enough for me to go on. I created a new issue to track this: https://github.com/bmd-studio/stm32-for-vscode/issues/74. The flags are not included in the generated makefile as of yet. Will try to release a hot fix today or tomorrow for this.

jortbmd, 2021-06-23:
Hi! The issue should have been fixed with the newest release. If you have any further issues please let me know.

bhatasaka, 2021-06-24:
> Hi! The issue should have been fixed with the newest release. If you have any further issues please let me know.

It works perfectly! You are amazing and so quick on this, thank you so much!

jortbmd, 2021-06-24:
Good to hear! It was one of the main features I wanted to tackle in the v3.0 release, so thank you for pointing it out!