bmd-studio/stm32-for-vscode #22: Advanced Makefile Application structure not supported

jorgeacortes, 2020-05-22:
When generating a project with the Advanced application structure (basically moves src and inc at root to a folder "Core") the project generation fails.

I guess problem is at this function checkForRequiredFiles(directoryFiles) but maybe there are other dependecies.

A note in readme could also be included meanwhile.

PS: Great project!

jortbmd, 2020-07-06:
Hi Jorge,

Thank you for the input! 

Sorry for the late reply. This issue did not pop on my radar.

It is true that the advanced feature is not supported. For now I do not see the benefit of supporting the advanced type of project, unless there is a good reason for using that structure (if you think there is let me know and I will consider adding using it). 

Will add it to the readme to clarify.

jortbmd, 2020-09-03:
Just to come back to this and update you. I have found that the new STM32CubeMX prefers the advanced structure and it is often grayed out and thus the Basic structure cannot be selected. So the option of doing both the Basic and Advanced structure has become relevant enough for me to address it as it would impact use-ability. For this reason I already pushed a new feature upstream which should allow the advanced structure to work. This will thus be included in the next release.

jortbmd, 2020-09-07:
The next release is out, so I am closing the issue and I hope it works well for you. Should you run into any issues let me know!

closed 2020-09-07 by jortbmd