bmd-studio/stm32-for-vscode #23: Noob Installation Instructions

G2Barbour, 2020-07-07:
Hello! I'm working on some TFT firmware for a 3D printer on an mkstft_28 clone with an STM32F107vct. For the past 3 months I have been manually debugging by hardcoding the display of variables during runtime. I recently ran into the limitations of such an approach with some very strange interrupt behavior, so I bought an ST Link v2 so I could use the on chip debugger. Which leads me here to your extension:

I am new to VS Code and when you say add a command line tool to the PATH, I have no idea what that means. 

When you say the project is initialized with CubeMX, I don't know what that means. 

And I don't know what the option to create a makefile is under Project Manager>Project>Toolchain IDE. 

I realize that for avid user this stuff is probably pretty basic, but I'm not as good with environments as I am with actually coding so I would appreciate any clarification you would be willing to provide.

seancsi, 2020-07-08:
We all started somewhere 😄 .
The PATH is something that is part of your operating system, so the instructions there depend on which one you're using.
In short, PATH is a common way to tell the command line where to find things in your system.  It's set by some installers automatically, but not all.  Some tools you'll need to use for development don't do this for you, or simply don't have installers.

On Windows 10, the easiest way to add something is to open the Start menu and search for "PATH".  It'll suggest "Edit the system environment variables."  Click that, and hit the "Environment Variables..." button in the bottom right.  You should see both User Variables and System Variables, both should have an entry in the list called "Path".  Highlight either of them, hit edit, and you should see a nice list of directories.  Hit Browse to add a directory of your choice.  In this case, you'll want to add the directory containing the executable you need in VS Code.

In order to build and debug, you probably need the following to work:
-[ARM GCC Compiler](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads)
-[GNUMake](http://gnuwin32.sourceforge.net/packages/make.htm)
-[OpenOCD](https://gnutoolchains.com/arm-eabi/openocd/)
-[STLink Drivers](https://www.st.com/en/development-tools/stsw-link009.html) or [Open Source](https://github.com/stlink-org/stlink)
-Cortex-Debug extension for VS Code (I think this extension prompts you to install it.)
Install everything, make sure everything is in your PATH, and you should be good to go.

CubeMX is a tool put out by ST that lets you configure the chip in a nice GUI and export a project template that's more or less ready to add your code to.  [It's available here.](https://www.st.com/en/development-tools/stm32cubemx.html)  Hopefully, your TFT firmware includes a CubeMX project file which would make re-generating everything really easy.  If it's custom code, CubeMX is a great thing to have at the beginning of your workflow, as it saves a lot of headaches, especially on complex projects with a lot of peripherals.

Hopefully that helps, and happy debugging!

G2Barbour, 2020-07-08:
Thank you for your speedy reply! I am going to try to follow these instructions immediately.

But I'm afraid I have already bricked the processor (at least temporarily). This has quickly turned into more than I tried to bite off. I just wanted to debug and now it won't boot. This project was my first experience with microprocessors at all, and I didn't even know what GPIO was 3 months ago. Now I'm neck deep in bootloaders. 

Unfortunately the project was built outside of CUBEMX, which I have now downloaded and installed. But I am able to connect with st link, which is how I cleared the read protection that allowed me to flash. From there, I went back into VS Code and after not completing the installation instructions above (as it was prior to your reply), I went into the normal debug process (F5 start debugging). I was very excited when I successfully connected! But I didn't get a good entry into the program, instead it halted the processor. I was thinking that using the debug method erased the bootloader, contrary to loading a bin with an SD card. I found the source code for something close to the original firmware, but I can't compile it because it doesn't have a platformio.ini file and that is the only way I have used VS code. I was also able to find what is allegedly a factory original bin for the mkstft_32, which is also very close in hardware. Close enough for the bootloader to be compatible was my hope. And it affected change! Instead of a deab black screen it started beeping. I expect that it was the result of naming real functions in the code after the weak ones in the bootloader and so I was able to successfully merge a few functions from the mks firmware to my BTT based source code and get a debug interrupt in actual code instead of an infinite loop of a single line in the bootloader. I was also pretty happy about that, because it meant I had a way to interact with it. I am thinking about hardocoding a way to clear whatever flag is set that prevents entering main(). The progess had me wondering if maybe I didn't actually erase the bootloader but got stuck in a debug mode or something. But the fact that it no longer flashed from the SD card has me concerned. I don't want to lose that functionality. 

I have metered out the board to find the boot0 pin and have a wire running to it through its on board capacitor dangling down by the st link. I have tried pulling it high to 3.3V, which basically shuts off the st link, and pulling it low which has no effect. 

I have also been able to read bits with the VS Code debugging terminal. Though I do not yet know how to change them. I can read the registers, but have almost no idea what they mean. I have very little exposure to assembly, I'm more of a c++ guy. I've always wanted to learn and I guess this is my chance, but these are not the best circumstances.

If I interrupt the process it stops in the function SPI_I2S_GetFlagStatus(), with bitstatus=reset if that means anything at all. The following bits are set high (I have only checked 2 registers): C_DEBUG, C_HALT, S_REGRDY, S_HALT, and TRCENA. The rest of the bits in both of those registers are 0. I am sure there are many other bits to look at but I was able to scrounge up some relevance for those with a few hours of Google.

I am going to start following the rest of your instructions now, but I am not sure if it will help me with the current situation. I know that your time is worth more than I can afford, but I will offer to pay you for your assistance in unbricking this thing. Would $50 by PayPal be insulting for you to touch this mess?

jortbmd, 2020-07-08:
Hi,

There is a lot to unpack in what you said and this is not really the place. I would advise you to take further questions somewhere else like the reddit page: https://www.reddit.com/r/stm32/. (I am also active there). 

Nonetheless I will do my best to give a good response. After which I am closing this issue as the above question is not relevant to the extension anymore. 

First off it is not clear if you are using firmware you yourself have written or are looking through the execution of pre existing firmware. If it is the latter it might be hard to actually see what is going on because you don't have the link with the source code, this would probably result in some weird unexpected behaviour if you tried debugging it with vscode. This could explain it not entering main and seeing an infinite loop on a single line.

The fact that it does not boot from the SD card is probably because you have overwritten the bootloader that was once on there. However this is probably fixable by flashing the original firmware on it using st-link (found something over here: https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT2.8-3.2).

I would also advise you to take a step back. You are working with a complex system which is new to you. I would advise you to leave the firmware you have for now and first create a simple blink program (turn a GPIO/LED on and off with a delay in between) using CubeMX and see if you can debug that.

Hope this helps.

closed 2020-07-08 by jortbmd

G2Barbour, 2020-07-10:
I know that you closed this issue and I don't mean to bother you, but that stm32 forum seems to be shut down. There hasn't been a post there in 4 months and I cannot post. Maybe because I haven't used Reddit since I started it over a year ago and do not have any reputation points or whatever they are called.

I was able to get my MCU unbricked. My theory is that when I click debug it only overwrites certain blocks of memory with the new code, and if I change it too much so that bits outside of that range shift I have to manually flash the compiled bin and then debug. Strange but it works. I was undwr the impression that debugging would automatically upload the whole program. Does your plugin seamlessly upload? I hadn't bothered pursuing it once I got it to work. 

I have also read that the stm32f107 has no bootloader that you can flash, which is great for me right now if it is true.

Anyway, I would really like to find somewhere, anywhere, that I can ask questions. Now that I can debug I have to attack the real problem, which is that the f_read function is returning data from the wrong sectors of the USB drive (the SD card works fine). I can't seem to find any good resources on how to debug the USB drive/file system relationship.

jortbmd, 2020-07-10:
Hi,

I do not think you have bricked your MCU. I just think that you are uploading the same binary file, while changing the code. So it does not show you the corresponding code. Mind that you have to first compile the code, then upload it and then debug it using a debugger. I do not know what your current setup is, but if you do not have those steps setup debugging will not actually show what you are doing. I would still advise you to first check if running a simple blinking led program and modifying that code works before attempting anything else. It is often a good idea to start with the basics. Also if you are not using STM32 for VSCode it is especially made to make the whole compile->flash->debug flow easier. Another thing you can look into is the [STM32CubdeIDE](https://www.st.com/en/development-tools/stm32cubeide.html). It is an intergrated development environment which comes with all the tools pre installed and might be a bit beginner friendly.

Also a bootloader is usually written itself, so almost all MCU's do not have a bootloader as a default, but it can be added. by uploading for example and USB bootloader.

For questions there is also the r/embedded or the r/STM32F103 forum. Also for resources, youtube, google, medium and stackoverflow are your friend.