bmd-studio/stm32-for-vscode #25: Buggy IntelliSense

robokamran, 2020-07-20:
Does anyone have a serious issue with the IntelliSense or is it just me? for example if I use uint32_t in my code, the whole code from that line will mess up. Yet it is ok with uint8_t :)) my `c_cpp_properties.json` for Microsoft C/C++ pligin is like this:

`
{
  "configurations": [
    {
      "name": "STM32",
      "includePath": [
        "Drivers/CMSIS/Device/ST/STM32F3xx/Include",
        "Drivers/CMSIS/Include",
        "Drivers/STM32F3xx_HAL_Driver/Inc",
        "Inc",
        "Drivers/STM32F3xx_HAL_Driver/Inc/Legacy"
      ],
      "defines": ["STM32F303xE", "USE_HAL_DRIVER"],
      "cStandard": "c99",
      "cppStandard": "c++11"
    }
  ],
  "version": 4
}
`

The code compiles, the case is with the IntelliSense. I know that some macros may be undefined for Microsoft C/C++ plugin, but is there any way that the code won't become like this? It slows down coding...

[image]

jortbmd, 2020-07-20:
Hi thanks for opening an issue.

I checked myself and I also get the same issue for the compilerPath. It seems that it needs an absolute path. Will try to include this in a next release.

Despite this it should still function. I do not know about the uint8_t vs uint32_t but when I include stdint.h all works as expected.
Did you include the stdint.h header?

robokamran, 2020-07-21:
Hi @jortbmd and thanks for noting this issue.
The project is generated by _STM32CubeMX_ and all the necessary headers are included in cascade (the code compiles successfully).
I have set an absolute path for `compilerPath` manually and the problem got fixed 👍. It seems that due to the nested includes and specific macros in them, the IntelliSense gets confused and it substantially needs the correct `compilerPath` to behave :))

jortbmd, 2020-09-07:
This is fixed in version 2.2.2 so I hope it helps and therefore I am closing this issue.

closed 2020-09-07 by jortbmd