bmd-studio/stm32-for-vscode #26: Easy user experience [Feature Request]

robokamran, 2020-07-21:
Hi! As you may know PlatformIO has support for STM32CubeMX either directly (partially) or through [stm32pio](https://github.com/ussserrr/stm32pio). It is a great tool but it is heavy and incomplete (and people don't like to use code transformers).
The current trend in real-time embedded development is an easy-to-use IDE with easy library management interface; something like _NuGet_ for _Visual Studio_ or libraries in _Arduino_  (as it is in PIO).

The idea of using CubeMX generated makefile is great and PIO doesn't have this and these items from PIO helps improving our UX:
- the arm-none-eabi-gcc is integrated in the plugin.
- the OpenOCD (from gnutoolchains.com) is also integrated in the plugin.

I don't know how it can happen, but the combination of PIO and stm32-for-vscode is tremendous and completely removes the need for IAR, Keil, etc.! Lightweight IDE, CubeMX supported, serial terminal, IntelliSense, easy installation and great library management system.

Moreover please add this link for [OpenOCD ](https://gnutoolchains.com/arm-eabi/openocd/) in the _marketplace_ page. It is very hard for unfamiliar users to find it.

Thank you in advance for your really cool efforts!

jortbmd, 2020-07-22:
Hi! Thanks for the interest and the recommendation. I added openOCD to the readme and it will be uploaded with the next release.

The PlatformIO not being able to deal with CubeMX files, and just really wanting an STM32 IDE in vscode were the reasons to make this whole extension. So good to hear I am not the only one! 

I acknowledge the value in integrating the build tools. I have not done it so far as I dislike the amount of openocd versions I already have on my system because of Eclipse based editors. However I do see value in integration of the build tools and will look into it and integrate it at point.

The library management and testing suite of PlatformIO is a definite plus, however it assumes the usage of Arduino, which I dislike for professional usage. For now integrating this is outside of the scope of this extension. However given enough time and donations I might look into rat least a library manager in the distant future (because I think this would be tremendously helpful for the whole embedded community as a whole).

For now I am super busy with a lot of refactoring and making sure the whole project has a continuous integration pipeline after which I hope I can do more frequent feature implementations. So it might take a while before I address the integration of the build tools, but it will come.

robokamran, 2020-07-23:
That's great, thanks again for all your efforts and it's all worthy!
If I one day learn how to JavaScript and typescript, I will get in the ring and give a hand ;)

klaygomes, 2020-08-07:
> For now I am super busy with a lot of refactoring and making sure the whole project has a continuous integration pipeline after which I hope I can do more frequent feature implementations. So it might take a while before I address the integration of the build tools, but it will come.

I may help with CI/CD integration.

jortbmd, 2020-08-13:
@klaygomes 
Thank you!
As stated in a different message I will merge everything I have right now so actual CI/CD integration might start. It would be amazing if you could contribute and very much appreciated.

klaygomes, 2020-08-14:
How could I contact you? My email is klaygomes at gmail dot com.

robokamran, 2020-09-15:
Thank you all very much!
Regarding the integration with PIO, I have also created an issue there to replace stm32pio unfamiliar interface with stm32-for-vscode. However, I have no additional information to add there. If you may find it helpful for the future of stm32-for-vscode, please add some comments on this [issue](https://github.com/platformio/platformio-vscode-ide/issues/1894).

jortbmd, 2021-04-13:
Automatic build tool installation is now part of version 3.0.0 which will be release very soon. So I am closing this issue.

closed 2021-04-13 by jortbmd