bmd-studio/stm32-for-vscode #29: Add external libraries

toledot21, 2020-08-13:
Hello,
I am fairly new with VSCode and your Extension (BTW: amazing extension, it helps me a LOT). Anyways,. I was wondering if you could let me know what is the correct procedure to compile external libraries.
Under your "HOW TO USE" section, you mentioned that external libraries should be added under the lib folder:
" Please note that external libraries should be placed in the lib folder and source code should be place in the src and inc folders".

Under my project, I have set up a Lib folder that contain my static library test.a.
on my Makefile I have:
# libraries -LLib
LIBS = -lc -lm -lnosys -ltest
LIBDIR = -LLib
LDFLAGS = $(MCU) -specs=nosys.specs -T$(LDSCRIPT) $(LIBDIR) $(LIBS) -Wl,-Map=$(BUILD_DIR)/$(TARGET).map,--cref -Wl,--gc-sections

However, when I run "Build Clean STM" and STM32Make.make is generated, it skips the library test.a and doesn't find the library directory. So on my terminal I get: undefined reference to 'foo_init'.

I would appreciate any help you might provide,

Regards,

jortbmd, 2020-08-13:
Hi,

Thank you for the kind words. This has been a bit of a premature readme update as the library function is as of yet not implemented. However I had somebody nice enough who already implemented it and did a pull request. For now it is a bit tedious to do it manually however it can be done. If you would like to. You can copy the STM32Make.make file rename it to something like: STM32MakeWithLibs  and add your libaries to the "LIBDIR = " line like so
``` 
LIBS = -lc -lm -lnosys 
LIBDIR = \
-LLib/test.a \
-LLib/someotherLibrary.a
```
.

Then you can run make on that makefile by invoking ```make -f STM32MakeWithLibs``` in the terminal. You should unfortunately redo this every time you add a file to your project.

I will see how fast I can make it go. I wanted a perfect release for the next version with everything fully tested, however it is taking me way to long to write all the testing, so I have been putting of merging and integrating new functionality. I will start on doing this very soon and will leave your issue open until then. If it is fixed I will close it so you will also get a notification once this is done.

toledot21, 2020-08-14:
Thank you for quick reply.
It worked, thanks!
I forgot to ask as well how can I configure the "DEBUG STM32" with this new  STM32MakeWithLibs?

robokamran, 2021-04-05:
Thanks for the solution, it worked for me. It would be very helpful if it is solved in the future. Because we use many static libraries and we love to use stm32-for-vscode!

robokamran, 2021-04-09:
Although, I have updated the original makefile generated by CubeMX, added static and other source libraries and include locations.  For static libraries I do this:
```
# libraries
LIBS = -lc -lm -lnosys 
LIBDIR = \
Middlewares/ST/STM32_MotionAC_Library/Lib/MotionAC_CM4F_wc32_ot_hard.a \
Middlewares/ST/STM32_MotionGC_Library/Lib/MotionGC_CM4F_wc32_ot_hard.a
```
and then I go to the project directory and simple enter just `make` and my project is easily built and there's no need to create another makefile for it. The good thing about it is that when I regenerate the project by CubeMX, all the changes I made to the makefile will remain and won't be removed. So I guess that maybe STM32-For-VSCODE extension won't even need to create another makefile for project compilation.
Another good thing about it will be that one can also add C++ sources to his/her own project (which is quite necessary these days).

jortbmd, 2021-04-13:
In the upcoming 3.0.0 version the issues mentioned here will be fixed. Libraries and library directories will be included from the makefile. On top of this additional libraries can be specified in the newly added project configuration file.

closed 2021-04-13 by jortbmd

robokamran, 2021-04-19:
Please consider that sometimes even the makefile generated by CubeMX is buggy. For example the makefile generated to include X-CUBE-MEMS-1 libraries does not work. The good thing about it is the edited makefile is not re-edited by CUBE-MX on regeneration of the project. So if STM32-For-VSCODE is strict to generate another makefile, please take that into consideration. Or else the developer must again use the terminal for build and program.

jortbmd, 2021-04-19:
Yes I am aware of the Makefile behaviour of STM32CubeMX and that it not always generates good makefiles. How the process works in v3+ is that it gets information from the CubeMX makefile and the configuration file (both which can be changed and will not be regenerated) and generate a resulting STM32Make file, so this should circumvent this issue. 
If you want to try V3 out it is now in a public beta. It can be installed trough a VSIX file, which you can find at the github releases page.