bmd-studio/stm32-for-vscode #30: Linker and make arguments

pooyasa, 2020-08-16:
Hi, 
First of all, I want to thank you for this amazing extension. Keil used to be a pain in the neck.
I found your extension today and I've been using it for couple of hours now, everything seems to work perfectly.
The problem is, I want to use C++ for programming ARM chips (STM32F103C8) and some of it's functionalities aren't working.
I can create a class and call it's methods and it works perfectly but I don't seem to be able to use std:string and try catch blocks.
I am using arm-none-eabi-gcc.exe.
Here is the Error:
`
undefined reference to `__gxx_personality_v0'
undefined reference to `__cxa_end_cleanup'
undefined reference to `__cxa_begin_catch'
`
And a few other more.
The other problem is that I want to use -j 8 argument in make but I don't now where I should put it.

jortbmd, 2020-08-18:
Hi,

Thank you for the kinds words. The standard c++ library is not included as it is rather large and would not fit on some smaller targets. However if you want to use it you can invoke the make command from the terminal as follows:
```make -f STM32Make.make LIBS="-lc -lm -lnosys -lstdc++"```
An option for adding different libraries in a config file will be added somewhere along the way, so then it should be easier. For now you could also add it to your tasks.json and as a new command (if copying the Build STM task replace: ```"type": "process",
			"command": "${command:stm32-for-vscode.build}"```, with ```"type": "shell", "command":"make -f STM32Make.make LIBS=\"-lc -lm -lnosys -lstdc++\""```

Will leave this issue open until I have added the library configuration option.

luckyspacetraveller, 2020-08-27:
Hi guys :)

actually one addition to this. The fix described works well for compiling, but e.g. including the <string> cpp standard library, in my case I still got an error.

This is due to the compiling goes with g++, but not the final linking.
In the file STM32Make.make, Line 215:

`$(CC) $(OBJECTS) $(LDFLAGS) -o $@`

should be changed to

`$(CXX) $(OBJECTS) $(LDFLAGS) -o $@`

when compiling with cpp files.

Cheers, and thanks for your amazing work!

jortbmd, 2020-08-31:
@luckyspacetraveller 
Hi!

Thanks for noting this and providing a fix. I will include it in the next release. I created a separate issue for this for ease of tracking. The new minor release will probably be somewhere this or next week so expect it to be included in there.

jortbmd, 2021-04-13:
Forgot to close this one. Already fixed in relation to issue #32

closed 2021-04-13 by jortbmd