bmd-studio/stm32-for-vscode #32: Not linking with g++ when compiling c++

jortbmd, 2020-08-31:
Hi guys :)

actually one addition to this. The fix described works well for compiling, but e.g. including the <string> cpp standard library, in my case I still got an error.

This is due to the compiling goes with g++, but not the final linking.
In the file STM32Make.make, around Line 215:

`$(CC) $(OBJECTS) $(LDFLAGS) -o $@`

should be changed to

`$(CXX) $(OBJECTS) $(LDFLAGS) -o $@`

when compiling with cpp files.

Cheers, and thanks for your amazing work!

_Originally posted by @luckyspacetraveller in https://github.com/bmd-studio/stm32-for-vscode/issues/30#issuecomment-682208167_

jortbmd, 2020-09-07:
This is fixed in the newest version so will close this issue.

closed 2020-09-07 by jortbmd

luckyspacetraveller, 2020-09-08:
@jortbmd Amazing work, thank you very much :)