bmd-studio/stm32-for-vscode #36: Make clean does not work on windows (command "rm" not found)

bw1faeh0, 2020-09-16:
When trying to make clean in a windows environment, terminal process ends with exit code 2, because command "rm" is not found:

````
> Executing task: make -f STM32Make.make clean <

cmd /c rm -rf build

Der Befehl "rm" ist entweder falsch geschrieben oder konnte nicht gefunden werden.
make: *** [clean] Fehler 1
Der Terminalprozess "C:\WINDOWS\System32\cmd.exe /d /c make -f STM32Make.make clean" wurde mit folgendem Exitcode beendet: 2.
````

Maybe `rd /s /q build` could be a solution.

jortbmd, 2020-09-16:
Hi! Thanks for the comment. Somehow I figured rm was standard in windows. Will look into it and will probably release a quick fix today or tomorrow.

jortbmd, 2020-09-16:
Just pushed the fix to the marketplace. This should resolve the issue. If not please let me know. For now I am closing this issue.

closed 2020-09-16 by jortbmd