bmd-studio/stm32-for-vscode #37: Code not working

itpalert, 2020-09-16:
Hi!
I have generated a project with CubeMx and exported it as makefile. I only selected the debug option to the SWD and to use the external crystal and configured the clocks.

My problem is that if I upload the code the crystal is not working. If I make the same project in CubeIde with the same settings but the CubeIde is generating the Makefile the crystal starts perfectly.

What Im missing? I attach the full project generated with CubeMx. I have the latest update for CubeMx. 

[pedo4.zip](https://github.com/bmd-studio/stm32-for-vscode/files/5233616/pedo4.zip)

jortbmd, 2020-09-16:
Thanks for opening an issue. I have looked into the different two makefiles, however the only difference I see is between the -specs used in the linker. Could you perhaps change: 
```LDFLAGS = $(MCU) -specs=nano.specs ....etc....``` to: ```LDFLAGS = $(MCU) -specs=nano.specs``` in the STM Makefile and see if that works?
You can then run it using the following command: ```make -f STM32Make.make``` after cleaning
I never had any issues with board containing crystals so I am really curious what causes this issue.

itpalert, 2020-09-16:
Hi!
I still dont have luck.
I had LDFLAGS = $(MCU) -specs=nosys.specs -T$(LDSCRIPT) $(LIBDIR) $(LIBS) -Wl,-Map=$(BUILD_DIR)/$(TARGET).map,--cref -Wl,--gc-sections and I changed it to LDFLAGS = $(MCU) -specs=nano.specs ... etc ...
I compiled the code with the command you wrote and flashed the chip with ST Utility. There is no signal from the chip, it doesnt run the code and it doesnt start the crystal.
If I program the chip from CubeIde dirrectly it works fine, I can debug it and everything. In this case the sine of the crystal is perfect. The exact type is STM32F103VCT6.
I didnt modified anything in the C codes, both projects have the same C code. Only the makefile is different.

itpalert, 2020-09-17:
Hi!
I found the cause of the problem. The CubeIde debugs the chip with stlink-dap.cfg file, but your plugin initialises the plain stlink.cfg file. The 'transport select "dapdirect_swd"' needs to be this, and it works fine.

jortbmd, 2020-09-28:
Hi, good to hear you resolved it yourself. I am leaving this issue open for now, as it is linked to me wanting a config file for openocd configuration and setup. As soon as this is done I will let you know so it will be easier to input a different configuration for the programmer.

jortbmd, 2021-04-13:
I have implemented a separate openocd.cfg file in version 3.0.0., which allows the user to use their own custom configurations. I have also created a way to change the programmer from a provided quickpick/dropdown list. For this reason I am marking this as resolved.

closed 2021-04-13 by jortbmd