bmd-studio/stm32-for-vscode #38: Publish on Open VSX Registry

lllearner90, 2020-09-25:
The extension looks great but it would be great if it can be made available to the open source versions of VS Code editor like VSCodium.
According to the VSCodium dev page, they use an open-source vendor-neutral registry OpenVSX to source extensions.
Publishing the extension there(OpenVSX) should be fairly simple and can be by running one command  `npx ovsx publish`

Kindly refer the  [issue](https://github.com/VSCodium/vscodium/issues/337#issuecomment-685731017) if you like to support this request.
Thanks.

jortbmd, 2020-09-28:
Hi,

Thank you for notifying me about this. I will look at it and if is not too much of a hassle (which from reading the issue you linked it would not be). I will publish it to the open source registry for the next version.

jortbmd, 2021-04-14:
Just a quick update. The extension has been published in the open VSX registry.

closed 2021-04-14 by jortbmd