bmd-studio/stm32-for-vscode #39: Ability to include custom source file path

lllearner90, 2020-10-02:
The extension places an limitation to place the application code in the Core/Inc & Core/Src path which sort of is a clutter (difficult to spot user code from the CubeMX generated code)
It will be great if the extension also supports an ability to add a custom path which can then be parsed to locate src/inc file to be included in the generated Makefile.
Thanks.

jortbmd, 2020-10-07:
Hi.

I am working on allowing settings per project, which will allow you to include folders and files using globs. However this might take a while before it is implemented. For now you can add any path you want to the original CubeMX makefile and it will be included when compiling.

lllearner90, 2020-10-09:
Thanks for the feedback @jortbmd, that is exactly how I am using the extension right now. Wasn't sure whether you had this in the pipeline so raised the issue.
Would have loved to contribute but sadly typescript is not something that either I have worked with or planning to work in near future.😅 Cheers!

jortbmd, 2021-04-13:
This has been added in version 3.0.0. Files can now be added in a separate configuration file, which also supports globs. For this reason I am closing this issue.3

closed 2021-04-13 by jortbmd