bmd-studio/stm32-for-vscode #41: On Windows 10

gmgunderground, 2020-10-17:
I followed all your meager tutorial, and on Windows 10 doesn't do noting, doesn't compile, no output no log of what is going wrong. Path to GCC is OK, path to make is ok. Very frustrating trying to make it work

jortbmd, 2020-10-17:
Hi,

If you have any recommendations or improvements for the tutorial I am happy to hear it. You can open a pull request or respond to this issue. 
I know installing tooling can be a pain an frustrating. There is a feature coming up where it tries to install all the requirements necessary. For now however, when the extension does not have the right paths for certain tools it will give a warning and a way to install them or enter the path. Check these messages to see what your are missing. You can also enter them when you go to file>preferences>settings and then under Extensions and STM32 for VSCode.
It is hard to debug your specific issue without any details so for now I will just give you suggestions on what you can watch out for and try.
- For windows be mindful of the direction of the slash. The path should use back slashes: \
- Double check if you are using wsl or something like fish. The paths should reflect the paths you would give the standard terminal in VSCode.
- Note that three tools need to be installed: gcc, make and openocd. If one of them is missing the extension wont work.
- Sometimes windows needs to restart for things added to path to be updated for all the terminal applications. So please restart Windows if you haven't already done so.
- If all fails I would advise you to add all the tools to PATH (you can do this by using the windows 10 search function next to start and search for: edit the system environment variables).

RicoKirstein, 2021-03-01:
I also encountered this bug.
the problem is:
return `GCC_PATH=${Helpers_1.convertToolPathToAbsolutePath(makeInfo.tools.armToolchainPath + '/arm-none-eabi-gcc', true)}`;
will create an invalid path and crash the build command without any output.
instead it should be 
return `GCC_PATH=${Helpers_1.convertToolPathToAbsolutePath(makeInfo.tools.armToolchainPath + path.sep + 'arm-none-eabi-gcc', true)}`;

00ihsan, 2021-03-02:
When I delete settings.json and reissue the command build and flash i get this as output in terminal:

> Executing task: make -f STM32Make.make flash <

openocd -f interface/stlink.cfg  -f target/stm32f3x.cfg -c "program build/LED.elf verify reset exit"
Open On-Chip Debugger 0.10.0 (2020-12-28) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J30M19 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.270417
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000c44 msp: 0x20010000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
Info : device id = 0x10036446
Info : flash size = 512kbytes
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
shutdown command invoked

Terminal will be reused by tasks, press any key to close it.

-------------------------------END------------------------------

After that, There is a new settings.json and the commands will not work.

The debugger gives this error :

[image]

jortbmd, 2021-04-13:
I have done extensive testing on the version 3.0.0 to prevent this. I have implemented space escaping and it now uses the posix standard for path separation. This should solve the issues you were having.

closed 2021-04-13 by jortbmd