bmd-studio/stm32-for-vscode #42: Debug error

shrey92, 2020-11-21:
I followed all the steps. I have added bin folders in my environment path (Windows 10). Then I made makefile with cubemx.
I can build the project using stm32 build. But, I cannot debug it. 
When I press F5 to debug, error pops up. "Failed to launch OpenOCD GDB Server: Error: spawn C:envoolspenOCD-20200729-0.10.0inENOENT" and there is an option to open launch.json.

I have tried to run openocd --version command in cmd and it returns the version, means it is added to path.

Please help me.

jortbmd, 2020-11-23:
Hi,

This sounds like it might be an issue with the path to openocd. Is it possible that you have two versions installed and you are using a different version for debugging. Check that you leave the path blank in settings->STM32 for VSCode->Open OCDPath when you have openOCD in your path. Also please check your .vscode/settings.json file which path is in there for openocd.

jortbmd, 2021-04-13:
Quick update on this. Build tools can now automatically be installed and the extension will check for them on start-up. This should prevent the issues you are having.

closed 2021-04-13 by jortbmd