bmd-studio/stm32-for-vscode #43: .cpp files doesn't appear in the STM32Make.make

Francois-Belanger, 2020-12-23:
Hi, I have added the following to the Makefile, but still the .cpp file doesn't appear in the STM32Make.make. 

CPP_SOURCES =  \
Core/Src/app_start.cpp

 Is there something I must do ?

Thanks

Francois-Belanger, 2020-12-24:
ok, must change main.c to main.cpp... but it doesn't work properly with cubeMx, as it always regain with main.c

closed 2020-12-24 by Francois-Belanger

jortbmd, 2020-12-24:
Hi,

That is true. I want to implement this at one point, however for now it something that has low priority at the moment, as cuyrrently there are some bigger features in the works.

The workaround I use when working with a C++ project is to create a seperate app() function in a seperate .h and .cpp file, which is placed in the main function in both the main.c file as the main.cpp file. When you regenerate you can then just copy paste the entire main.c file into the main.cpp file without running into issues with user code. The main.c file will not be included  while compiling when a main.cpp file is present so you don't have to delete it. I know it is not ideal and I will let you know when the syncing is setup however this might take some time.

robin7331, 2021-04-08:
Is there any update on this? 😋