bmd-studio/stm32-for-vscode #44: launch.json debug config doesn't work

dljsjr, 2020-12-31:
### Environment Info
Visual Studio Code Version: 1.52.1
stm32-for-vscode Version: 2.2.4
cortex-debug Version: 0.3.7
OS: macOS Big Sur 11.1 (Intel)

### Bug Info
When running the built-in Build or Flash tasks, a launch.json gets generated for running debug configs. This config includes `"preLaunchTask": "Build STM",`, which is the offending line.

When running the debugger w/ this line present, VS Code times out after a few seconds with the following pop-up:

[image]

Unfortunately, since there is no option to just truck on, VS Code kills the running task and the debugger never attaches.

Commenting out that line allows for debugging to proceed appropriately.

jortbmd, 2021-01-22:
Hi,

My apologies for the late reply. This is some behaviour I have not seen before. The Build STM task should be able to be tracked. I could advise you to delete the task.json file and let it be generated again. If this does not fix the issue please let me know so I can look into it.

dljsjr, 2021-01-26:
Hey there! Sorry myself for the late response! I tried that and it didn't seem to work. The only way to get it to work is to comment out that line and make sure I build by hand before debugging.

jortbmd, 2021-01-26:
No worries,

Could you perhaps share your launch.json file and your task.json file? Or even better a project in which the error is reproducable? Does this happen on a specific project or various projects? Would be great if I had a way to produce the issue myself.

jortbmd, 2021-03-02:
I have run into the issue myself now. This is due to an update of VSCode itself and the way it handles tasks. There currently is a standard output for non tracked tasked. Tracking seems to just be declaring what the end of stdout needs to be for when the tasks is complete. I first will finish up some major refactoring and feature additions and than I will look into this.

dljsjr, 2021-03-02:
@jortbmd That's good news, I totally forgot to send you my tasks.json and I apologize!

jortbmd, 2021-04-13:
I think I have fixed the problem in version 3.0.0. There was an issue were the build task would not complete (an unresolved promise). This has been fixed and I haven't encountered the tracking issue any more. Should this pop up again feel free to open another issue. For now I am closing it.

closed 2021-04-13 by jortbmd