bmd-studio/stm32-for-vscode #46: OpenGCD not recognizing the NUCLEO board

NathanWeasley, 2021-01-27:
Hi :
I have everything the extension requires(arm-gcc, cortex-debug, make, official stlink driver), and already built a project generated by STM32CubeMX, which was successful, and was trying to download it onto a NUCLEO board, that's when the following happened:

1. The first time I download, the program was burnt into the chip, and I was able to see serial prints on a console.
2. After the successful download, the debug session DID NOT start, reporting there was an error, without any details.
3. And I tried to hit that download-and-debug command again, and from this time on, it NEVER even started to download.

The error print was: 

> openocd -f interface/stlink.cfg  -f target/stm32f1x.cfg -c "program build/P2.5_64x32_F103.elf verify reset exit"
> Open On-Chip Debugger 0.10.0 (2020-12-28) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> Info : clock speed 1000 kHz
> Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
> Info : STLINK V2J30M19 (API v2) VID:PID 0483:374B
> Info : Target voltage: 3.251613
> Error: init mode failed (unable to connect to the target)  
> in procedure 'program'
> ** OpenOCD init failed **
> shutdown command invoked

I've tried to switch USB ports, and searched through a dozen similar problems but none helped. I am hoping nice developers of this extension to help me.

Thanks.

jortbmd, 2021-01-27:
Hi!

Well this (hopefully) nice developer has run into a similar issue. First off the f1 has multiple bootmodes which can be set using the boot0 and boot1 pins, so watch out that you use the correct ones, however I assume that they will be already set correctly on the nucleo board. Just in case there is more info here: https://electronics.stackexchange.com/questions/82336/whats-the-usage-of-the-three-boot-mode-of-stm32f103

Besides this you need to set the debugging  mode. This can be done using CubeMX in System Core>Sys>Debug and then set it to Serial Wire. Regenerate your files, program your chip and then it should work. If this fails build everything again and try with the Cubeprogrammer to erase the mcu and upload your current firmware, which can be found in the build directory (should be build/P2.5_64x32_F103.elf in your case).

Hope this helps!

NathanWeasley, 2021-01-27:
I figured it out on your help, thanks, you are of great help.

I guess the key hint is that "Succeeded the first time and never again" thing.
I was stupid enough to alter the initial settings of debug interfaces to SWD, in a TOTALLY WRONG way. It appears STM32CubeMX use JTAG as default debug interface, I am new to this code generator and I was like "cool I will disable additional JTAG pins so that I can have more GPIOs with only SWDIO and SWCLK left", totally unaware of that such alternation will cause the code generator to inteprate such action as "no debug port AT ALL".
I grounded BOOT0 and regenerated the code and now I have my board back.

Again, thanks a lot for your quick reply.

closed 2021-01-27 by NathanWeasley