bmd-studio/stm32-for-vscode #49: Nothing working anymore

00ihsan, 2021-03-02:
I installed all the prequired [rpgra,s amd tried tu build and flash. It asked for the path to OCD,  I gave it. everything worked ok except the debugger. when i tried to fix the debugger, it looked like it was working but i couldn't use the ctrl + shift + p commands.

Please help because i need this for school.

jortbmd, 2021-03-02:
Hi,
You could try looking at the debug configuration for the Cortex Debug extension, because that is basically the  extension which handles all the debugging. STM32 for VSCode only build the binary file, which Cortex Debug uses for debugging. I would advise you to check the settings for Cortex Debug. As far as I know the extension should not be able to freeze VSCode in that manner so it might be something else that is causing your issue.

00ihsan, 2021-03-02:
Hi,

I have it working now. But when I want to debug, the debugger just spins and spins and spins.

[image]

jortbmd, 2021-03-03:
Hello,

The debugging flow consists basically of build->flash->debug, usually when the debuggers gets stuck it means that it either doesn't build or flash properly. I would advise you to check if your build completes. You can see this in the terminal. On top of that if you go to the debug console, you can also see if flashing works and that openocd is operating nicely.

It is always hard debugging issues like yours. If you can post your task.json, launch.json and settings.json files from the .vscode directory I might be able to spot a problem there.

jortbmd, 2021-04-13:
Closing this issue due to inactivity. As a small adittion a lot of the flows have been revisited in version 3.0.0. This issue might have been due to the build not completing in a correct manner. This has been fixed in version 3.0.0.

closed 2021-04-13 by jortbmd