bmd-studio/stm32-for-vscode #50: Add some buttons at the bottom

cccbug, 2021-03-18:
Hello,
If it possible to add such as build debug & flash button looks like 
[image]

There is source code，https://github.com/espressif/vscode-esp-idf-extension

Thanks.

jortband, 2021-03-18:
Hi,
I was already working on something similar. For the next big update that I am finishing up I wanted to put the build and flash commands in the sidebar. As can be seen below. The idea is that I will also extend this menu further with some additional features after the current major update has been finished. Hope this is close enough to what you are looking for.
[image]

cccbug, 2021-03-18:
Ok，Thank you

closed 2021-03-18 by cccbug