bmd-studio/stm32-for-vscode #55: GNU Arm Embedded Toolchain Escaped Path for 3.0.0 on Windows

mattparks, 2021-04-14:
I updated the version of STM32 for VSCode to 3.0.0, I'm now unable to build on my Windows 10 machine.

[image]

```sh
D:/PROGRAM\ FILES\ (X86)/GNU\ ARM\ EMBEDDED\ TOOLCHAIN/10\ 2020-Q4-MAJOR/BIN/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32WB35xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32WBxx/Include -IDrivers/CMSIS/Include -IDrivers/STM32WBxx_HAL_Driver/Inc -IDrivers/STM32WBxx_HAL_Driver/Inc/Legacy -IMiddlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_WPAN -IMiddlewares/ST/STM32_WPAN/ble -IMiddlewares/ST/STM32_WPAN/ble/core -IMiddlewares/ST/STM32_WPAN/ble/core/auto -IMiddlewares/ST/STM32_WPAN/ble/core/template -IMiddlewares/ST/STM32_WPAN/ble/svc/Inc -IMiddlewares/ST/STM32_WPAN/ble/svc/Src -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/shci -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/tl -IMiddlewares/ST/STM32_WPAN/utilities -ISTM32_WPAN/App -IUSB_Device/App -IUSB_Device/Target -IUtilities/lpm/tiny_lpm -IUtilities/sequencer -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ADS131M06.d" -Wa,-a,-ad,-alms=build/ADS131M06.lst Core/Src/ADS131M06.c -o build/ADS131M06.o
'D:/PROGRAM\' is not recognized as an internal or external command,
operable program or batch file.
make: *** [build/ADS131M06.o] Error 1
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command make -f STM32Make.make" failed to launch (exit code: 1).
```

jortbmd, 2021-04-14:
Hi!

Thank you for raising this issue. I have found the bug and will release a hotfix somewhat later today. I will keep you posted.

jortbmd, 2021-04-14:
The hotfix  (v3.0.1) is in. You can install this and now you should be able to build with the settings you have.

mattparks, 2021-04-14:
Now running v3.0.1, I'm still unable to build on Windows. I think STM32 for VSCode might be finding the path for Make and OpenOCD and overriding my VS Code settings because on every reload the paths change back to uppercase. Even when pointing to the correct one I get the same error:
```sh
> Executing task: "D:\PROGRAM FILES (X86)\GNUWIN32\BIN\MAKE.EXE" -f STM32Make.make <

x86 : The term 'x86' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify 
that the path is correct and try again.
At line:1 char:19
+ D:\PROGRAM FILES (X86)\GNUWIN32\BIN\MAKE.EXE -f STM32Make.make
+                   ~~~
    + CategoryInfo          : ObjectNotFound: (x86:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
 
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command "D:\PROGRAM FILES (X86)\GNUWIN32\BIN\MAKE.EXE" -f STM32Make.make" terminated with exit code: 1.
```

jortbmd, 2021-04-14:
Behavior for powershell and command differ on this point. Quoting stops came from parsing the path, while powershell still tries and trips over the x86 part. I will look at this tomorrow and I will probably release have a fix for it during the day.

jortbmd, 2021-04-15:
Hi,

I created a fix for it in V3.0.2 I tested it on my machine and this seams to resolve it. If you see the chance please test it for yourself and reopen this issue again if any issues persist.

mattparks, 2021-04-19:
I updated to v3.0.4 on Windows, I'm getting path errors in the build process:

```sh
> Executing task: & \"D:\PROGRAM FILES (X86)\GNUWIN32\BIN\MAKE.EXE\" -j16 -f STM32Make.make  <

mkdir build
D:/PROGRAM\ FILES\ (X86)/GNU\ ARM\ EMBEDDED\ TOOLCHAIN/10\ 2020-Q4-MAJOR/BIN/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32WB35xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32WBxx/Include -IDrivers/CMSIS/Include -IDrivers/STM32WBxx_HAL_Driver/Inc -IDrivers/STM32WBxx_HAL_Driver/Inc/Legacy -IMiddlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_WPAN -IMiddlewares/ST/STM32_WPAN/ble -IMiddlewares/ST/STM32_WPAN/ble/core -IMiddlewares/ST/STM32_WPAN/ble/core/auto -IMiddlewares/ST/STM32_WPAN/ble/core/template -IMiddlewares/ST/STM32_WPAN/ble/svc/Inc -IMiddlewares/ST/STM32_WPAN/ble/svc/Src -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/shci -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/tl -IMiddlewares/ST/STM32_WPAN/utilities -ISTM32_WPAN/App -IUSB_Device/App -IUSB_Device/Target -IUtilities/lpm/tiny_lpm -IUtilities/sequencer -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/ADS131M06.d" -Wa,-a,-ad,-alms=build/ADS131M06.lst Core/Src/ADS131M06.c -o build/ADS131M06.o
D:/PROGRAM\ FILES\ (X86)/GNU\ ARM\ EMBEDDED\ TOOLCHAIN/10\ 2020-Q4-MAJOR/BIN/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32WB35xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32WBxx/Include -IDrivers/CMSIS/Include -IDrivers/STM32WBxx_HAL_Driver/Inc -IDrivers/STM32WBxx_HAL_Driver/Inc/Legacy -IMiddlewares/ST/STM32_US'D:/PROGRAM\' is not recognized as an internal or external command,
operable program or batch file.
B_Device_Library/Class/CDC/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_WPAN -IMiddlewares/ST/STM32_WPAN/ble -IMiddlewares/ST/STM32_WPAN/ble/core -IMiddlewares/ST/STM32_WPAN/ble/core/auto -IMiddlewares/ST/STM32_WPAN/ble/core/template -IMiddlewares/ST/STM32_WPAN/ble/svc/Inc -IMiddlewares/ST/STM32_WPAN/ble/svc/Src -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/shci -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/tl -IMiddlewares/ST/STM32_WPAN/utilities -ISTM32_WPAN/App -IUSB_Device/App -IUSB_Device/Target -IUtilities/lpm/tiny_lpm -IUtilities/sequencer -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/app_debug.d" -Wa,-a,-ad,-alms=build/app_debug.lst Core/Src/app_debug.c -o build/app_debug.o
D:/PROGRAM\ FILES\ (X86)/GNU\ ARM\ EMBEDDED\ TOOLCHAIN/10\ 2020-Q4-MAJOR/BIN/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32WB35xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32WBxx/Include -IDrivers/CMSIS/Include -IDrivers/STM32WBxx_HAL_Driver/Inc -IDrivers/STM32WBxx_HAL_Driver/Inc/Legacy -IMiddlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_WPAN -IMiddlewa'D:/PROGRAM\' is not recognized as an internal or external command,
operable program or batch file.
res/ST/STM32_WPAN/ble -IMiddlewares/ST/STM32_WPAN/ble/core -IMiddlewares/ST/STM32_WPAN/ble/core/auto -IMiddlewares/ST/STM32_WPAN/ble/core/template -IMiddlewares/ST/STM32_WPAN/ble/svc/Inc -IMiddlewares/ST/STM32_WPAN/ble/svc/Src -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/shci -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/tl -IMiddlewares/ST/STM32_WPAN/utilities -ISTM32_WPAN/App -IUSB_Device/App -IUSB_Device/Target -IUtilities/lpm/tiny_lpm -IUtilities/sequencer -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/app_entry.d" -Wa,-a,-ad,-alms=build/app_entry.lst Core/Src/app_entry.c -o 
build/app_entry.o
D:/PROGRAM\ FILES\ (X86)/GNU\ ARM\ EMBEDDED\ TOOLCHAIN/10\ 2020-Q4-MAJOR/BIN/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32WB35xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32WBxx/Include -IDrivers/CMSIS/Include -IDrivers/STM32WBxx_HAL_Driver/Inc -IDrivers/STM32WBxx_HAL_Driver/Inc/Legacy -IMiddlew'D:/PROGRAM\' is not recognized as an internal or external command,
operable program or batch file.
ares/ST/STM32_USB_Device_Library/Class/CDC/Inc -IMiddlewares/ST/STM32_USB_Device_Library/Core/Inc -IMiddlewares/ST/STM32_WPAN -IMiddlewares/ST/STM32_WPAN/ble -IMiddlewares/ST/STM32_WPAN/ble/core -IMiddlewares/ST/STM32_WPAN/ble/core/auto -IMiddlewares/ST/STM32_WPAN/ble/core/template -IMiddlewares/ST/STM32_WPAN/ble/svc/Inc -IMiddlewares/ST/STM32_WPAN/ble/svc/Src -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/shci -IMiddlewares/ST/STM32_WPAN/interface/patterns/ble_thread/tl -IMiddlewares/ST/STM32_WPAN/utilities -ISTM32_WPAN/App -IUSB_Device/App -IUSB_Device/Target -IUtilities/lpm/tiny_lpm -IUtilities/sequencer -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/hw_timerserver.d" -Wa,-a,-ad,-alms=build/hw_timerserver.lst Core/Src/hw_timerserver.c -o build/hw_timerserver.o  
[... cut ...]
MAKE.EXE: *** [build/stm32wbxx_hal_adc.o] Error 1
MAKE.EXE: *** [build/stm32wbxx_hal_adc_ex.o] Error 1
MAKE.EXE: *** [build/stm32wbxx_hal_cortex.o] Error 1
MAKE.EXE: *** [build/stm32wbxx_hal_dma.o] Error 1
MAKE.EXE: *** [build/stm32wbxx_hal_dma_ex.o] Error 1
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command & \"D:\PROGRAM FILES (X86)\GNUWIN32\BIN\MAKE.EXE\" -j16 -f STM32Make.make " terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

jortbmd, 2021-04-20:
Hi this seems like an issue with powershell, escaping and the way the arm-none-eabi path is done in the makefile. Will reopen this issue and will look for a fix.

jortbmd, 2021-04-22:
I have fixed this in the new v3.0.5 release. I have tested this for Make, OpenOCD and the ARM toolchain. So I am hoping this issue is finally resolved fully. For now I will close it again.

closed 2021-04-22 by jortbmd