bmd-studio/stm32-for-vscode #59: Faster compilation using -j flag?

TomN7, 2021-04-15:
Is it possible to speed up compilation using the make -j flag to build multiple targets in parallel?
Compared to building in STM32CubeMX, using VSCode is significantly slower and I think this is the cause.

jortbmd, 2021-04-15:
Hi!

You are absolutely right. I checked it against one of my projects and it went from 37 seconds to 12 seconds when using this flag. I have added this as a standard in the new v3.0.2. release.

closed 2021-04-15 by jortbmd

TomN7, 2021-04-15:
Cheers, looking forward to it. It would probably be useful to be able to set the maximum number of parallel tasks in the config file as “-j” on its own is unlimited and can freeze the PC (see https://unix.stackexchange.com/questions/316644/is-make-j-with-no-argument-dangerous).