bmd-studio/stm32-for-vscode #69: An error when building a project with space symbol in name

sinkapoy, 2021-05-13:
If project target contains " ", make gives something like
`make: *** No rule to make target '<last word of the target>.hex', needed by 'all'.  Stop.`
`make: *** Waiting for unfinished jobs....`
`The terminal process "/bin/bash '-c', '"make" -j16 -f STM32Make.make '" terminated with exit code: 2.`

jortbmd, 2021-05-17:
Hi. I will look into this and come up with a quick fix when I have time this week. For now the solution would be to change the target name in the STM32-for-VSCode.config.yaml file to something without a space.

jortbmd, 2021-05-22:
This has been fixed in the new v3.0.7 version. You need to delete your launch.json and task.json files and let them regenerate for it to work.

closed 2021-05-22 by jortbmd