bmd-studio/stm32-for-vscode #71: Doesn't seem to launch

doubleopinter, 2021-05-21:
Hey I don't know how if this is a bug or not but I need a bit of help. After I install the extension I don't get the icon in the side bar, it's like it's just not running. I'm on macOS and I have an M1 mac.

jortbmd, 2021-05-21:
It does not always show, only when there is a .ioc (stm32 cube file) or a STM32-for-VSCode.config.yaml file in the workspace. However you can run the openExtension command to open the extension and show the icon. You can do this by pressing cmd+shift+p and searching for: "Open the STM32 for VSCode extension".

It might also have something to do with the M1 mac variant, however I do not see a direct reason for this.

doubleopinter, 2021-05-21:
Thanks for the comment, I’ll try to give it a go tonight and let you know.

closed 2021-05-26 by doubleopinter