bmd-studio/stm32-for-vscode #75: Can not build touchGFX projects

Jona2892, 2021-06-28:
Hey there,

is there any way to build and flash stm32-projects including touchGFX? Has anyone done this before?
I can build any stm32-project created as makefile-project from cubeMX, but not with touchGFX.

Jonathan

jortbmd, 2021-06-29:
Hi Jonathan,

I had a quick look. It looks like the generated code form CubeMX with the touchGFX package isn't that great and needs some configuration. First of TouchGFX is .cpp based so you have to rename or copy and rename you main.c file into a main.cpp file.

On top of that the generated TouchGFX folder is not included in the makefile as a set of files that need to be compiled. Your best option is to add this in the STM32-forVSCode.config.yaml. e.g.:
```
includeDirectories: 
  - Inc/**
  - Core/Inc/**
  - Core/Lib/**
  - TouchGFX/**

sourceFiles:  
  - Src/**
  - Core/Src/**
  - Core/Lib/**
  - TouchGFX/**
```
On top of this you need to generate some files using TouchGFX however I do not have direct access at this very moment to a Windows computer, so I cannot check if this generates all the required files and it compiles. Hopefully this already helps. Let me know if you have further issues.

Jona2892, 2021-06-30:
Hey,

thanks for your answer! I will give it a try in the next days and will report here :)

Best
Jonathan

jortbmd, 2021-07-27:
> Hey,
> 
> thanks for your answer! I will give it a try in the next days and will report here :)
> 
> Best
> Jonathan

Do you by any chance have an update? Would be interesting to hear if this has worked out.

jortbmd, 2021-11-18:
Closed due to inactivity.

closed 2021-11-18 by jortbmd