bmd-studio/stm32-for-vscode #76: ERROR: No Link found for this specific platform - possible M1 issue?

A6Deg, 2021-07-09:
Hey guys,

i get this error when I try to install the build tools. I followed the guide and installed:
gcc
make and
openocd 
and copied the path to the settings.json file but I get the error:

"Something has gone wrong while installing the build toold: Error: No link found for this specific platform, please open a new issue to ask for support for your specific platform at: https://github.com/bmd-studio/stm32-for-vscode/issues"

I checked all the installations with "--version" flag and they are all installed... 

Does this have anything to do with the M1 processor i am using? Is Macbook pro M1 supported?

Regards,
Almog

A6Deg, 2021-07-11:
Ok, Managed to get the build tools installed (I think)

I have on the left hand side: 
[image]
 
But now I am encountering new errors:
1. I have a lot of undefined in the source code

[image]
"variable uint32_t is not a type name"
This error is for all uint*_t types.

2. When I try and build the basic example I get this error:
[image]

Are these errors related? What am I missing?
Any help is appreciated!

Regards, 
Almog

jortbmd, 2021-07-11:
Hi!
Thank you for opening up this issue. When making the build tool installer the M1 was still too fresh for some of the packages. I will roll out a quick fix today which should resolve this issue. However I am not certain it will work as I cannot test on an M1 macbook.

It seems you have added the correct paths to the build tools. The context menu would not show up otherwise. At first build most of the error will go away as the intellisense will be configured upon first build.

For the issue of the uin32_t types, it is as simple as including the stdint library in your file e.g.:
```
#include <stdint.h>
```

Hope this helps!

A6Deg, 2021-07-11:
Will this patch fix the error I'm having? ("TypeError: Cannot read property 'split' of NULL")

I am happy to test it out for you once it's out and let you know!

Almog

jortbmd, 2021-07-11:
This patch will probably fix the red line issues when you build once again, as this is still intellisense that is not configured correctly.

For the typerror this very much seems like an issue in your code, where you try to acces a property named split from a null pointer. Which means you probably have a pointer to a struct in which has the split property, but the pointer isn't referencing the actual object. The error will probably give you a line number and file, where you can look for the issue. If the issue still persists I would recommend checking out stack overflow for a possible solution.

P.S. I am closing the issue as the initial feature request has been resolved.

closed 2021-07-11 by jortbmd

A6Deg, 2021-07-18:
Just wanted to update - works on M1. 
Able to build and flash. 
Great work, thanks!

gupengzu, 2024-11-26:
hei,i also meet the same problem .can you help me?
[image]