bmd-studio/stm32-for-vscode #78: Something went wrong during the build process: TypeError: Cannot read property 'split' of null

ad7718, 2021-07-27:
Generated default F103 makefile project with cubeMx. Getting error "Something went wrong during the build process: TypeError: Cannot read property 'split' of null" when trying to build. VScode 1.58.2.

jortbmd, 2021-07-27:
Hi!

That sounds more like a issue in the code itself, rather than an issue of the extension. However it might be related to it not including build files or specific libraries. Could you tell me what the default F103 project is? If you could share your CubeMX file that would be even better. Also on which OS are you running VSCode?

ad7718, 2021-07-27:
Hi. Tried on MX linux 19.4 and on Windows 10. IOC file in attachment(just selected stm32f103c8t6 and used settings in readme). 
[stm32f103.ioc.zip](https://github.com/bmd-studio/stm32-for-vscode/files/6885199/stm32f103.ioc.zip)

jortbmd, 2021-07-27:
Hi again. Just tried it and it seems to build fine on my Windows 10 machine.
It maybe has something to do with your paths. Could you indicate what the path to your folder is and where your build tools are located?

ad7718, 2021-07-27:
I don think this is related to path, build tools was downloaded by your extension on both systems, no spaces, English only:
"/home/user/Desktop/stm32f103/"  "/usr/bin/make /home/user/.config/VSCodium/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools". I also tested on VScodium and VScode.

ad7718, 2021-07-27:
I've figured out what's happened. Steps to reproduce: press Build on empty folder(or folder without Makefile?), empty .yaml is generated, generate cubeMxCode, press Build again and you'll get error. And when I'm deleting .yaml, it's regenerated and build is successfull.

jortbmd, 2021-07-31:
Hi good for you to mention. I will add a warning if no makefile type project is present, and perhaps some setup/configuration steps afterwards to solve this issue.

jortbmd, 2021-11-18:
Fixed with commit: https://github.com/bmd-studio/stm32-for-vscode/commit/b561eccba72605afa19801fee5132234e2e478e3

closed 2021-11-18 by jortbmd