bmd-studio/stm32-for-vscode #79: An error when building a project with space symbol in make.exe location

alient12, 2021-07-27:
Location of make.exe has " " so building a project will failed.
```bash
> Executing task: "C:\USERS\ALI ENT\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\WINDOWS-BUILD-TOOLS\4.2.1-2.1\.CONTENT\BIN\MAKE.EXE" -j16 -f STM32Make.make  <

'C:\USERS\ALI' is not recognized as an internal or external command,
operable program or batch file.
The terminal process "C:\WINDOWS\System32\cmd.exe /d /c "C:\USERS\ALI ENT\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\WINDOWS-BUILD-TOOLS\4.2.1-2.1\.CONTENT\BIN\MAKE.EXE" -j16 -f STM32Make.make " terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
stm-for-vscode version: 3.1.4
OS: Windows_NT x64 10.0.19043
vscode version: 1.58.2

jortbmd, 2021-07-27:
Hi. It shouldn't do that and I do not have similar behavior when using cmd. Could you perhaps tell me which automation shell and which regular shell you are using?

alient12, 2021-07-27:
Hi. Both of them are CMD. Also, I put this line in the settings.json to sure it uses CMD.
```json
"terminal.integrated.automationShell.windows": "CMD"
```
After running Build button in extension, This will show to me

[image]

jortbmd, 2021-07-29:
That is weird. As far as I can tell this should have worked, as it is encapsulated by the quotes. If you use the integrated terminal and type in:
```
"C:\USERS\ALI ENT\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\WINDOWS-BUILD-TOOLS\4.2.1-2.1\.CONTENT\BIN\MAKE.EXE" -j16 -f STM32Make.make
```
Does it give you the same error?

alient12, 2021-08-02:
Thanks for your response. It worked but gave me a stm32f1xx_hal library error.
```bash
C:\Users\ali entezari\Documents\STM Projects\STM32F103C8T6\NRF24_Reciever_makefile>"C:\USERS\ALI ENTEZARI\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\WINDOWS-BUILD-TOOLS\4.2.1-2.1\.CONTENT\BIN\MAKE.EXE" -j16 -f STM32Make.make
compilation terminated.MAKE.EXE: *** [STM32Make.make:167: build/main.o] Error 1In file included from Core/Inc/spi.h:28,
                 from Core/Src/spi.c:21:Core/Inc/main.h:31:10: fatal error: stm32f1xx_hal.h: No such file or directory   31 | #include "stm32f1xx_hal.h"
      |          ^~~~~~~~~~~~~~~~~compilation terminated.MAKE.EXE: *** [STM32Make.make:167: build/spi.o] Error 1
```
Although the header files are in the directory

[image]

jortbmd, 2021-08-02:
That is a weird error to have. Is it possible that you generated and then tried to build the project when on a different project type e.g. EWARM? I can advice you to throw away the STM32-for_VSCode.config.yaml file, the makefile. Then regenerate with CubeMX and do a clean build afterwards.
If that does not help please share your Makefile and STM32-for-VSCode.config.yaml file, so I can track down the problem

alient12, 2021-08-02:
Thanks. I deleted STM32-for-VSCode.config.yaml and Makefile and then regenerated them with MXCube. Clean Build gave me previous error but manually typing ```"C:\USERS\...\BIN\MAKE.EXE" -j16 -f STM32Make.make clean``` and  ```"C:\USERS\...\BIN\MAKE.EXE" -j16 -f STM32Make.make``` worked great without any error.

jortbmd, 2021-08-12:
Good te hear! Has the path issue been resolved in the latest version of STM32 for VSCode? If not I still have to do some investigating, why this is the case for you. If this does not work you can also edit you tasks.json file to automate this for now. E.g. add this to tasks 
```
		{
			"label": "Build STM32",
			"type": "shell",
			"command": "C:\USERS\...\BIN\MAKE.EXE" -j16 -f STM32Make.make",
			"options": {
				"cwd": "${workspaceRoot}"
			},
			"group": {
				"kind": "build",
				"isDefault": true
			},
			"problemMatcher": [
				"$gcc"
			],
			"dependsOn": [
				"Build STM"
			]
		},
```
Of course you need to replace: C:\USERS\...\BIN\MAKE.EXE" -j16 -f STM32Make.make with the correct path.

alient12, 2021-08-12:
I've changed tasks.json, and it didn't affect, still same error. I have to run the command manually. 
I believe it might be the vscode bug. 

> Has the path issue been resolved in the latest version of STM32 for VSCode?

I've tested 3.1.4 and 3.1.5 beta, but none of them worked.

> Of course you need to replace: C:\USERS...\BIN\MAKE.EXE" -j16 -f STM32Make.make with the correct path.

Of course, I did.

alient12, 2021-08-12:
> I believe it might be the vscode bug.

closed 2021-08-12 by alient12