bmd-studio/stm32-for-vscode #80: Build Issue

natankeddem, 2021-07-29:
I can't seem to get build to work but I think I have an idea why but not how to fix it.  I get this terminal output:
```
> Executing task: & \"F:\DOWNLOADS\MAKE-3.81-BIN\BIN\MAKE.EXE\" -j16 -f STM32Make.make  <

The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & \"F:\DOWNLOADS\MAKE-3.81-BIN\BIN\MAKE.EXE\" -j16 -f STM32Make.make " terminated with exit code: 1.
```
Note the "\" after make.exe. These are my extension setting paths:
```
  "stm32-for-vscode.armToolchainPath": "C:\\Program Files (x86)\\GNU Tools Arm Embedded\\7 2018-q2-update\\bin",
  "stm32-for-vscode.openOCDPath": "F:\\DOWNLOADS\\OPENOCD-20210625\\OPENOCD-20210625-0.11.0\\BIN\\OPENOCD.EXE",
  "stm32-for-vscode.makePath": "F:\\DOWNLOADS\\MAKE-3.81-BIN\\BIN\\MAKE.EXE"
```
VS Code : 1.58.2
OS:  Windows_NT x64 10.0.19043
stm32-for-vscode: v3.1.4

natankeddem, 2021-07-29:
The gnu Make binary .zip doesn't seem to include all the required libraries so I ended up just installing the complete "Setup" version.

jortbmd, 2021-07-29:
Did you fix it by yourself? You can also let the extension install all the required build tools for you to get a more streamlined experience.

natankeddem, 2021-07-29:
I am not sure what happened but the automatic install didn't work for me, I needed to manually install things.  Manually using the "Setup" make installer fixed the issue for me.

jortbmd, 2021-07-29:
Good to hear that you fixed this. Do you have any idea why the setup might have failed? Otherwise I need to provide a fix for this.

natankeddem, 2021-07-29:
I uninstalled everything to investigate just that since flashing\debugging wasn't working after I got the build stuff sorted out.  Here are my logs from "Install Build Tools":
```
> Executing task: & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64\npx\" xpm install --global @xpack-dev-tools/openocd@latest <

Terminal will be reused by tasks, press any key to close it.

> Executing task: & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64\npx\" xpm install --global @xpack-dev-tools/windows-build-tools@latest <

Terminal will be reused by tasks, press any key to close it.

> Executing task: & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64\npx\" xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest <

Terminal will be reused by tasks, press any key to close it.
```

I get no errors and wait until the last prompt for awaiting install.  Where would these files install?  The path settings never get updated to anything, they just stay false.

natankeddem, 2021-07-29:
Looking further into xpm; I have never used it in the past.  It appears like it was not properly installed?  Therefore it could not install the rest of the items?
I did:
`C:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64>npm install --global xpm@latest`
Which seemed to allow the other items to install correctly.

jortbmd, 2021-07-29:
That should install the build tools. Which should be under 
```
c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools
```
However it will probably spawn a separate window to install them if you are using powershell. It might take some time for all of them to complete and it might look finished once it is done. If nothing changes after installing you can press check build tools. It should then automatically find the build tools if all are installed.

natankeddem, 2021-07-29:
The initial setup of xpm seems to have failed, I needed to manually run:
`C:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64>npm install --global xpm@latest`
Once I manually ran that the rest properly installs.

natankeddem, 2021-07-29:
When installing the build tools if I don't run that command manually I get this error in the vs code log:
```
[2021-07-29 14:34:30.306] [renderer1] [error] Error: EBUSY: resource busy or locked, rmdir 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64': Unknown (FileSystemError) (FileSystemError): Error: EBUSY: resource busy or locked, rmdir 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.5.0-win-x64'
	at _handleError (c:\Users\nkeddem\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:84:168274)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
```

jortbmd, 2021-07-30:
Thanks for checking! This helps I am guessing it is because it tries to run several things in parallel, and because it does not know when the separate tasks complete it tries to clean up once it is done, however as the resource is still being used it throws the error you show me. This should be fixable and will get on it!

natankeddem, 2021-07-30:
Sounds good, let me know if you want me to try anything on my end.  If i knew where any other logs were located I would give them to you but I didn't see anything obvious.

jortbmd, 2021-07-31:
Thanks for the help with testing. I think I have devised a solution. If you could test the beta release that would be great. You can find it over here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.1.5-beta
To install this open VSCode. Press ctrl+shift+p and search for Extensions: Install from VSIX, then you can you select the file and it should install it.

natankeddem, 2021-08-02:
Some progress, at least it is erroring out now...
```
> Executing task: & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd\" xpm install --global @xpack-dev-tools/openocd@latest <

'xpm' is not recognized as an internal or external command,
operable program or batch file.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd\" xpm install --global @xpack-dev-tools/openocd@latest" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
xpm isn't getting installed, see folder contents:
[image]

jortbmd, 2021-08-02:
Powershell: the shell that keeps on giving ;). So xpm should not be installed. It is run by using npx, which allows you to cache and execute an npm package without installing it. Xpm itself is a package installer for the build tools and these should in turn be installed in: c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools.
When using powershell on my setup I do not get the same behavior, which makes debugging a bit harder.
Could you perhaps try and run the commands below in powershell in VSCode? If it works, it will ask you if you want to install openocd but you can just exit or press n. I am only curious if this runs xpm.
Do note that I am assuming you still have the node instance in the "c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node\node-v16.6.0-win-x64" folder.
```& 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest```

```& \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd\" xpm install --global @xpack-dev-tools/openocd@latest```

natankeddem, 2021-08-02:
I am guessing this is what you wanted? 

```
PS C:\Stm32Ws\g47test0> & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest
'xpm' is not recognized as an internal or external command,
operable program or batch file.
PS C:\Stm32Ws\g47test0> & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd\" xpm install --global @xpack-dev-tools/openocd@latest
& : The term '\c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.6.0-win-x64\npx.cmd\' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was 
included, verify that the path is correct and try again.
At line:1 char:3
+ & \"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm3 ...
+   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (\c:\Users\nkedd...in-x64\npx.cmd\:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

jortbmd, 2021-08-12:
Yes. Thanks! My apologies for the late response. I did not have time to dive further into this. As of yet I cannot replicate the same behavior you are having so it is a bit hard to debug.  I will investigate further and report back if I have a potential fix.

jortbmd, 2022-02-21:
Hi! I have been doing some integration testing to make sure the whole installation flow works on clean systems. I have fixed some potential bugs in there. I would greatly appreciate if you can test out the installation one more time.

natankeddem, 2022-02-21:
> Hi! I have been doing some integration testing to make sure the whole installation flow works on clean systems. I have fixed some potential bugs in there. I would greatly appreciate if you can test out the installation one more time.

What version do you want me to try? The last release I see was in November.

jortbmd, 2022-02-21:
Hi. Thanks for the reply! I jumped the gun on  posting this and did not publish the newest version right away. However it is published now. It is version 3.2.0

natankeddem, 2022-02-21:
I see that new version now, I can't seem to get the extension to show up in the sidebar. I feel like I am missing something simple here but I can't get it to show up.

natankeddem, 2022-02-21:
Forgot about having to have a project open. Install fails with following:
```
> Executing task: & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v17.5.0-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/openocd@latest <

'xpm' is not recognized as an internal or external command,
operable program or batch file.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v17.5.0-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/openocd@latest" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

jortbmd, 2022-02-21:
Thank you for checking. There still seems to be an issue then. In that case I will keep on looking. Thank you for checking!

jortbmd, 2025-01-20:
Just to double check if this issue still persists. I have updated the install mechanism a few times the past couple of versions. Could you check if the issue still persists?

natankeddem, 2025-01-20:
Wow, a blast from the past, I can try to but I need to get spooled back up to what was going on here. I seem to have purged out most of my knowledge.

natankeddem, 2025-01-20:
As I said it has been a while since I have touch this stuff. I tried my best to purge out previous installs from AppData before installing. I think this was a success?
```bash
 *  Executing task: & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.20.2-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/openocd@latest 

npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'xpm@0.20.5',
npm WARN EBADENGINE   required: { node: ' >=18.0.0' },
npm WARN EBADENGINE   current: { node: 'v16.20.2', npm: '8.19.4' }
npm WARN EBADENGINE }
@xpack-dev-tools/openocd@0.12.0-4.1...
@xpack-dev-tools/openocd@0.12.0-4.1 => 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\openocd\0.12.0-4.1'
Downloading https://github.com/xpack-dev-tools/openocd-xpack/releases/download/v0.12.0-4/xpack-openocd-0.12.0-4-win32-x64.zip...
Extracting 'xpack-openocd-0.12.0-4-win32-x64.zip'...
1113 files => 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\openocd\0.12.0-4.1\.content'
npm notice 
npm notice New major version of npm available! 8.19.4 -> 11.0.0      
npm notice Changelog: https://github.com/npm/cli/releases/tag/v11.0.0
npm notice Run npm install -g npm@11.0.0 to update!
npm notice 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.20.2-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/windows-build-tools@latest 

@xpack-dev-tools/windows-build-tools@4.4.1-3.1...
@xpack-dev-tools/windows-build-tools@4.4.1-3.1 => 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\windows-build-tools\4.4.1-3.1'
Downloading https://github.com/xpack-dev-tools/windows-build-tools-xpack/releases/download/v4.4.1-3/xpack-windows-build-tools-4.4.1-3-win32-x64.zip...
Extracting 'xpack-windows-build-tools-4.4.1-3-win32-x64.zip'...
40 files => 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\windows-build-tools\4.4.1-3.1\.content'
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task: & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\node\node-v16.20.2-win-x64\npx.cmd' xpm install --global @xpack-dev-tools/arm-none-eabi-gcc@latest 

@xpack-dev-tools/arm-none-eabi-gcc@14.2.1-1.1.1...
@xpack-dev-tools/arm-none-eabi-gcc@14.2.1-1.1.1 => 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\arm-none-eabi-gcc\14.2.1-1.1.1'
Downloading https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/download/v14.2.1-1.1/xpack-arm-none-eabi-gcc-14.2.1-1.1-win32-x64.zip...
Extracting 'xpack-arm-none-eabi-gcc-14.2.1-1.1-win32-x64.zip'...
6210 files => 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\arm-none-eabi-gcc\14.2.1-1.1.1\.content'
```

I am not sure if it is just the random new project I created but it doesn't seem to build correctly? This was just me picking a random board and exporting everything at default...
```bash
 *  Executing task: & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\windows-build-tools\4.4.1-3.1\.content\bin\make.EXE' -j16 DEBUG=1 -f STM32Make.make  

"c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\arm-none-eabi-gcc\14.2.1-1.1.1\.content\bin/arm-none-eabi-gcc" @build/debug/asdf.elf.in -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Wl,--print-memory-usage -specs=nano.specs  -TSTM32G473QETx_FLASH.ld  -lc -lm -lnosys  -Wl,-Map=build/asdf.map,--cref -Wl,--gc-sections -o build/debug/asdf.elf
c:/Users/nkeddem/AppData/Roaming/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/14.2.1-1.1.1/.content/bin/../lib/gcc/arm-none-eabi/14.2.1/../../../../arm-none-eabi/bin/ld.exe:STM32G473QETx_FLASH.ld:56: 
syntax error
collect2.exe: error: ld returned 1 exit status
make.EXE: *** [STM32Make.make:315: build/debug/asdf.elf] Error 1

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'c:\Users\nkeddem\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\windows-build-tools\4.4.1-3.1\.content\bin\make.EXE' -j16 DEBUG=1 -f STM32Make.make " terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

jortbmd, 2025-01-23:
Thansk for checking. The build error does seem to originate from the linkerscript of the mcu and not the application. However if it is an all defaults project it should build with no issue. Will try to reproduce with the same chip if I have the time to double check where the issue might originate from.