bmd-studio/stm32-for-vscode #81: VSCode settings pollution

pepijnverburg, 2021-08-24:
The VSCode project settings of projects where this extension is NOT being used are being polluted with some automatic initialization of the settings for the extension? 

Perhaps it is better practise when they are not set, they are not added to the project settings, but just have an internal fallback?

These two settings are added all the time in random projects:
```
{
  "stm32-for-vscode.openOCDPath": false,
  "stm32-for-vscode.armToolchainPath": false
}
```

jortbmd, 2021-09-10:
Hi Pepijn,

This was done with an old version, however this should have been resolved in v3. The only time it updates the local path is when a non valid path is given in the current workspace settings for the STM32-for-VSCode tooling, and this is only done when using the extension, or upon activation of the extension, which it only does automatically when a .ioc file or a STM32-for-VSCode.config.yaml file is present.
I pasted the code for V3 workspace settings management below (can be found in src/buildtools/index.ts:47). This only should allow to write a path to the workspace settings when one is already present and it is a valid path.
```
  // check if there is a local settings file and update if necessary.
  // NOTE: settings should not be added or edited in the workspace for STM32 for vscode,
  // however old version of STM32 for VSCode did put it in the workspace folder
  const localUpdatePromises: Thenable<void>[] = [];
  if (vscode?.workspace?.workspaceFolders?.[0]) {
    const localSettingsPath = path.join(vscode.workspace.workspaceFolders[0].uri.fsPath, '.vscode', 'settings.json');

    const localExtensionSettings =
      vscode.workspace.getConfiguration('stm32-for-vscode', vscode.Uri.file(localSettingsPath));
    _.forEach(finalBuildTools, (toolPath, key) => {
      
      const localPath = localExtensionSettings.get(key);
      if (localPath === undefined) { return; }
      let localWhichPath = which(localPath);
      if (key === 'armToolchainPath') {
        localWhichPath = which(path.join(`${localPath}`, 'arm-none-eabi-gcc'));
      }
      if ((!_.isEqual(toolPath, localPath) && !localWhichPath) || !localPath) {
        localUpdatePromises.push(localExtensionSettings.update(key, toolPath, vscode.ConfigurationTarget.Workspace));
      }
    });
  }
  ```

Should the issue persist let me know. However my guess is this is stems from the older version.

thernstig, 2023-11-04:
@jortbmd is it easy for you to fix this? It is problematic when working many projects.

jortbmd, 2023-11-06:
@thernstig thanks for confirming this is an issue. This has been on the back-burner for a while. Theoretically it should be easy to fix and in theory the addition should only happen when a build in STM32-for-VSCode is started. I currently do not have a way to reproduce this bug. If you have a project where this happens with and if you know when it happens I could dive into it a little bit more to see when these settings are generated and come up with fix probably pretty fast.

jortbmd, 2024-01-09:
Hi all,

This issue should be resolved for now in the 3.2.6 release. Thank you @cvanbeek13 so much for the fix!
For this reason I am closing this issue. If it should pop up again feel free to re-open this issue or create a new one.

closed 2024-01-09 by jortbmd