bmd-studio/stm32-for-vscode #82: Problem with GDB Server on Cortex-Debug...

robokamran, 2021-09-10:
I get this error `Waiting for gdb server to start...` on the console when I use `stm32-for-vscode` extension. Everything works fine (compile, program, etc.), aside that Cortex-Debug always is waiting for some unknown to me GDB server to start!

[image]

When I try to kill the terminal it says `gdb-server terminal closed unexpectedly. Trying to reopen it`.

[image]

Is this a problem with `stm32-for-vscode` extension or `Cortex-Debug`?

jortbmd, 2021-09-10:
Hi,

I have the same behavior, which is related to cortex debug. It seems like a task is running in the background to automatically connect to a gdb-server as soon as one starts, however for some reason it is opened in the terminal as a task, which it did not do before. I also find it mildly annoying because my terminal always jumps back to the gdb-server instance.

closed 2021-09-10 by jortbmd

robokamran, 2021-09-10:
Thanks for swift reply @jortbmd 
Yeah the terminal always jumping back to the gdb-server stuff is pretty much annoying. The stm32-for-vscode works clean and just fine yet this issue make things a bit dirty. I'll try to open an issue in Cortex-Debug.

jortbmd, 2021-09-10:
Opening an issue at Cortex-Debug would be great. I had a quick look at their code and the whole terminal seems a bit superfluous as it seems to only serve the purpose of communicating back to the user, which is also done in the output tab. Also there would be no reason, why they couldn't just start up the terminal as soon as a debugging session starts. Do not I have looked at the code for 5 min, so there might be more going on.

robokamran, 2021-09-27:
Hi @jortbmd I've not seen the problem on any of my computers since I posted the issue on Cortex-Debug, do you still see this problem on your console?