bmd-studio/stm32-for-vscode #83: No rule to make target if using "Build" or "Flash STM32" tasks.

3IMVS, 2021-09-25:
Hello, loving your extension so far, although I have faced an issue. 
I'm trying to work on some F4 DSP example project's and just can't make it work. I feel like my lack of makefile knowledge is the reason why i can't troubleshoot this myself. 

Problem:
When using "Clean build" task everything builds and links successfully, but when later on trying to use flash or build tasks makefile produces:
`MAKE.EXE: *** No rule to make target drivers\bsp\components\common\audio.h', needed by build/main.o.  Stop.
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command & \"C:\PROGRAM FILES (X86)\GNUWIN32\BIN\MAKE.EXE\" -j16 -f STM32Make.make " terminated with exit code: 1.`

audio.h is used by cs43l22.h. 

makefile:
```make
##########################################################################################################################
# File automatically-generated by tool: [projectgenerator] version: [3.14.1] date: [Sat Sep 25 13:59:08 EEST 2021] 
##########################################################################################################################

# ------------------------------------------------
# Generic Makefile (based on gcc)
#
# ChangeLog :
#	2017-02-10 - Several enhancements + project update mode
#   2015-07-22 - first version
# ------------------------------------------------

######################################
# target
######################################
TARGET = lab1

######################################
# building variables
######################################
# debug build?
DEBUG = 1
# optimization
OPT = -Og

#######################################
# paths
#######################################
# Build path
BUILD_DIR = build

######################################
# source
######################################
# C sources
C_SOURCES =  \
Core/Src/main.c \
Core/Src/gpio.c \
Core/Src/stm32f4xx_it.c \
Core/Src/stm32f4xx_hal_msp.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ramfunc.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr_ex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.c \
Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_exti.c \
Core/Src/system_stm32f4xx.c \
Core/Src/dma.c \
Core/Src/i2s.c \
[... 451 lines cut ...]
  - Core/Src/**
  - Core/Lib/**
  - Drivers/CMSIS/DSP/Include 
  - Drivers/BSP/STM32F411E-Discovery 
  - Drivers/BSP/PDM/Inc

# Files that should be included in the compilation.
# If a CubeMX makefile is present it will automatically include the  c and cpp/cxx files from that makefile.
# Glob patterns (https://www.wikiwand.com/en/Glob_(programming)) can be used.
# Do mind that double stars are reserved in yaml
# these should be escaped with a: \ or the name should be in double quotes e.g. "HARDWARE_DRIVER*.c"
sourceFiles:  
  - Src/**
  - Core/Src/**
  - Core/Lib/**
  - Drivers/BSP/STM32F411E-Discovery/stm32f411e_discovery_audio.c 
  - Drivers/BSP/Components/cs43l22/cs43l22.c 
  - Drivers/BSP/STM32F411E-Discovery/stm32f411e_discovery.c 

# When no makefile is present it will show a warning pop-up.
# However when compilation without the CubeMX Makefile is desired, this can be turned of.
suppressMakefileWarning: false
    
```
[image]

jortbmd, 2021-09-27:
Hi,

Thanks that it is working out for you. From what you are sending me it seems that the board support package (BSP) is not included on compilation. I do not see it in the CubeMX makefile. I do not know if you have enabled this in CubeMX? If so deleting the current makefile and regenerating the project might fix this issue. If this does not help you can add the files to the filelist in the STM32-for-VSCode.config.yaml file. Under sourceFiles and includeDirectories. It might take you a few compilation cycles to track down every file you need to use. Or you can just add Drivers/BSP/** to include all the files in the folder and check if that works.

Hope this helps!

3IMVS, 2021-09-27:
Hello once again, thank you very much for your thorough answer. You've helped me to troubleshoot this, I've tried everything you've suggested, but the culprit was make version. I've ran make -f STM32Make -d and observed something interesting:
```
No need to remake target `Drivers/BSP/STM32F411E-Discovery/../Components/cs43l22/cs43l22.h'.
      Considering target file `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h'.
       File `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h' does not exist.
       Looking for an implicit rule for `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h'.
       Trying pattern rule with stem `audio.h'.
       Trying implicit prerequisite `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h,v'.
       Trying pattern rule with stem `audio.h'.
       Trying implicit prerequisite `d\:\code\ssa\lab1\drivers\bsp\components\common\RCS/audio.h,v'.
       Trying pattern rule with stem `audio.h'.
       Trying implicit prerequisite `d\:\code\ssa\lab1\drivers\bsp\components\common\RCS/audio.h'.
       Trying pattern rule with stem `audio.h'.
       Trying implicit prerequisite `d\:\code\ssa\lab1\drivers\bsp\components\common\s.audio.h'.
       Trying pattern rule with stem `audio.h'.
       Trying implicit prerequisite `d\:\code\ssa\lab1\drivers\bsp\components\common\SCCS/s.audio.h'.
       No implicit rule found for `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h'.
       Finished prerequisites of target file `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h'.
      Must remake target `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h'.
make: *** No rule to make target `d\:\code\ssa\lab1\drivers\bsp\components\common\audio.h', needed by `build/stm32f411e_discovery_audio.o'.  Stop.
``` 
For every other file make uses relative path, but audio.h somehow behaves differently, as you can see there is an extra slash after partition letter ```( d\:\ ) ```. I don't know why this happened specifically to audio.h file, but I've uninstalled make, and installed a newer version of make through chocolatey. And straight away everything worked like a charm.

Additional info if someone might stumble on this issue:
I was running this setup on windows and somehow installed an older version of make:
```
make -v
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
```
after installing with chocolatey:
```
make -v
GNU Make 4.3
Built for Windows32
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Also, STM32Cube doesn't have an option to include BSP drivers for cs43l22 specifically for STM32F411E-DISCO, or at least i was unable to find them, that's why i had to copy them manually.

closed 2021-09-27 by 3IMVS