bmd-studio/stm32-for-vscode #86: Can not find non standard library

ayleenw, 2021-11-09:
I try to tell stm32-for-vscode to use a non standard library, that I have copied into /Core/Lib, by adding the library incl. the path to the libraries section in the config.yaml file:

libraries: 
  - c
  - m
  - nosys
  - C:/Users/xxx/AppData/Local/Notebook.rep/stm32code/xxx/Core/Lib/libCOM.a

but the linker gives me this error message:
"../arm-none-eabi/bin/ld.exe: cannot find -lC:/Users/xxx/AppData/Local/Notebook.rep/stm32code/xxx/Core/Lib/libCOM.a"

The file exists at this path. I also tried the relative path, I tried putting the path in quotes, I tried backslashes ... to no avail. How should I enter the lib location correctly?

Thanks for your great work - it is so much better to use VSCode over Eclipse ^^

jortbmd, 2021-11-09:
The library location should be put in the libraryDirectories option. I see that I haven't commented this piece in the config file. Will do this in the future. However the easy fix should be to include the directory of the library in the libraryDirectories option e.g.:
```
# libraries to be included. The -l prefix to the library will be automatically added.
# Mind that non standard libraries should have a path to their respective directory.
libraries:
  - c
  - m
  - nosys
 - libCOM # if this does not work try ibCOM gcc has a weird thing with parsing the starting l for library names

libraryDirectories:
 - C:/Users/xxx/AppData/Local/Notebook.rep/stm32code/xxx/Core/Lib
```

Let me know if that works!

ayleenw, 2021-11-10:
Thanks for coming back to me so quickly. I had even tried to put the relative path into the [] braces that were in the config file, but given up on that too early. After some testing I found the solution now (remembering, that for CubeIDE I had to rename COM.a to libCOM.a, but using COM as the name inside the Settings window):

`
libraries: 
  - c
  - m
  - nosys
  - COM

libraryDirectories: [C:/Users/xxx/AppData/Local/Notebook.rep/stm32code/xxx/Core/Lib/]
`

Thanks again for your great work and the heads up!

jortbmd, 2021-11-11:
Good to hear that it has been fixed now! Will leave the issue open for now, as this is a good reminder to add more comments to the config file concerning libraries and library directories.

jortbmd, 2021-11-18:
Closing this issue as some extra comments to the yaml config file concerning libary directories, which will be updated in the next release.

closed 2021-11-18 by jortbmd