bmd-studio/stm32-for-vscode #87: Linker error when using TouchGFX library

mikenorgate, 2021-11-15:
I am getting the following errors when attempting to build 

```
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/GeneratedFont.o:(.rodata._ZTIN8touchgfx13GeneratedFontE+0x8): undefined reference to `typeinfo for touchgfx::ConstFont'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/FrontendApplicationBase.o:(.rodata._ZTIN8touchgfx14MVPApplicationE[_ZTIN8touchgfx14MVPApplicationE]+0x8): undefined reference to `typeinfo for touchgfx::Application'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/RatingHeaderBase.o:(.rodata._ZTIN8touchgfx6WidgetE[_ZTIN8touchgfx6WidgetE]+0x8): undefined reference to `typeinfo for touchgfx::Drawable'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/RatingHeaderBase.o:(.rodata._ZTI16RatingHeaderBase+0x8): undefined reference to `typeinfo for touchgfx::Container'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/SensorContainerBase.o:(.rodata._ZTIN8touchgfx20AbstractGraphElementE[_ZTIN8touchgfx20AbstractGraphElementE]+0x8): undefined reference to `typeinfo for touchgfx::CanvasWidget'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/SensorContainerBase.o:(.rodata._ZTIN8touchgfx11GraphScrollILs60EEE[_ZTIN8touchgfx11GraphScrollILs60EEE]+0x8): undefined reference to `typeinfo for touchgfx::DataGraphScroll'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/SensorContainerBase.o:(.rodata._ZTI19SensorContainerBase+0x8): undefined reference to `typeinfo for touchgfx::Container'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/sensor_display_screenViewBase.o:(.rodata._ZTIN8touchgfx4ViewI30sensor_display_screenPresenterEE[_ZTIN8touchgfx4ViewI30sensor_display_screenPresenterEE]+0x8): undefined reference to `typeinfo for touchgfx::Screen'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/TouchGFXConfiguration.o:(.rodata._ZTIN8touchgfx9PartitionINS_4meta8TypeListI30sensor_display_screenPresenterNS2_INS1_3NilES4_EEEELt1EEE[_ZTIN8touchgfx9PartitionINS_4meta8TypeListI30sensor_display_screenPresenterNS2_INS1_3NilES4_EEEELt1EEE]+0x8): undefined reference to `typeinfo for touchgfx::AbstractPartition'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/TouchGFXConfiguration.o:(.rodata._ZTIN8touchgfx9PartitionINS_4meta8TypeListI25sensor_display_screenViewNS2_INS1_3NilES4_EEEELt1EEE[_ZTIN8touchgfx9PartitionINS_4meta8TypeListI25sensor_display_screenViewNS2_INS1_3NilES4_EEEELt1EEE]+0x8): undefined reference to `typeinfo for touchgfx::AbstractPartition'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/TouchGFXConfiguration.o:(.rodata._ZTIN8touchgfx9PartitionINS_4meta8TypeListINS_12NoTransitionENS2_INS1_3NilES4_EEEELt1EEE[_ZTIN8touchgfx9PartitionINS_4meta8TypeListINS_12NoTransitionENS2_INS1_3NilES4_EEEELt1EEE]+0x8): undefined reference to `typeinfo for touchgfx::AbstractPartition'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/TouchGFXConfiguration.o:(.rodata._ZTIN8touchgfx5NoDMAE[_ZTIN8touchgfx5NoDMAE]+0x8): undefined reference to `typeinfo for touchgfx::DMA_Interface'
c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: build/TouchGFXGeneratedHAL.o:(.rodata._ZTI20TouchGFXGeneratedHAL+0x8): undefined reference to `typeinfo for touchgfx::HAL'
```

The various TouchGFX types it is complaining about are included in the `libtouchgfx-float-abi-hard.a` library

My `STM32-for-VSCode.config.yaml` looks like this:

```
# Configuration file for the STM32 for VSCode extension
# Arrays can be inputted in two ways. One is: [entry_1, entry_2, ..., entry_final]
# or by adding an indented list below the variable name e.g.:
# VARIABLE:
# - entry_1
# - entry_2

# The project name
target: NUCLEO-WB55
# Can be C or C++
language: C++

optimization: Og

# MCU settings
targetMCU: 
cpu: -mcpu=cortex-m4
fpu: -mfpu=fpv4-sp-d16
floatAbi: -mfloat-abi=hard
ldscript: NUCLEO-WB55/STM32CubeIDE/STM32WB55RGVX_FLASH.ld

# Compiler definitions. The -D prefix for the compiler will be automatically added.
cDefinitions:
 - USE_HAL_DRIVER
 - STM32WB55xx
 - DEBUG
cxxDefinitions:
 - USE_HAL_DRIVER
 - STM32WB55xx
 - DEBUG
asDefinitions: []

# Compiler definition files. you can add a single files or an array of files for different definitions.
# The file is expected to have a definition each new line.
# This allows to include for example a .definition file which can be ignored in git and can contain
# This can be convenient for passing along secrets at compile time, or generating a file for per device setup.
cDefinitionsFile:
cxxDefinitionsFile:
asDefinitionsFile:

# Compiler flags
cFlags: []
cxxFlags: []
assemblyFlags: 
 - -specs=nano.specs

# libraries to be included. The -l prefix to the library will be automatically added.
# Mind that non standard libraries should have a path to their respective directory.
libraries:
 - m
 - stdc++
 - supc++
 - :libtouchgfx-float-abi-hard.a
libraryDirectories:
 - "Middlewares/ST/touchgfx/lib/core/cortex_m4f/gcc"

# Files or folders that will be excluded from compilation.
# Glob patterns (https://www.wikiwand.com/en/Glob_(programming)) can be used.
# Do mind that double stars are reserved in yaml
# these should be escaped with a: \ or the name should be in double quotes e.g. "**.test.**"
excludes: 
  - "Middlewares/ST/touchgfx/framework/source/platform/hal/simulator/sdl2/HALSDL2.cpp"
  - "Middlewares/ST/touchgfx/framework/source/platform/hal/simulator/sdl2/HALSDL2_icon.cpp"
  - "Middlewares/ST/touchgfx/framework/source/platform/hal/simulator/sdl2/OSWrappers.cpp"
  - "Middlewares/ST/touchgfx/os/OSWrappers_cmsis.cpp"
  - "Middlewares/ST/touchgfx/os/OSWrappers.cpp"
  - "Middlewares/ST/touchgfx/framework/source/platform/driver/touch/SDL2TouchController.cpp"
  - "Middlewares/ST/touchgfx/framework/source/touchgfx/widgets/**"
  - "Middlewares/ST/touchgfx/framework/source/touchgfx/containers/**"
  - "TouchGFX/generated/simulator/src/mainBase.cpp"
  - "TouchGFX/simulator/main.cpp"

# Include directories (directories containing .h or .hpp files)
# If a CubeMX makefile is present it will automatically include the include directories from that makefile.
includeDirectories: 
  - "NUCLEO-WB55/Drivers/STM32WBxx_HAL_Driver/Inc"
  - "NUCLEO-WB55/Drivers/STM32WBxx_HAL_Driver/Inc/Legacy"
  - "NUCLEO-WB55/Drivers/CMSIS/Device/ST/STM32WBxx/Include"
  - "NUCLEO-WB55/Core/Inc"
  - "Core/Inc"
  - "Middlewares/Third_Party/FreeRTOS/Source/include"
  - "Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2"
  - "Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F"
  - "Middlewares/Third_Party/ARM_CMSIS/Core/Include"
  - "Middlewares/Third_Party/ARM_CMSIS/RTOS2/Include"
  - "Middlewares/ST/touchgfx/framework/include"
  - "TouchGFX/generated/fonts/include"
  - "TouchGFX/generated/gui_generated/include"
  - "TouchGFX/generated/images/include"
  - "TouchGFX/generated/texts/include"
  - "TouchGFX/gui/include"
  - "TouchGFX/App"
  - "TouchGFX/target"
  - "Displays"
  - "Sensor"

# Files that should be included in the compilation.
# If a CubeMX makefile is present it will automatically include the  c and cpp/cxx files from that makefile.
# Glob patterns (https://www.wikiwand.com/en/Glob_(programming)) can be used.
# Do mind that double stars are reserved in yaml
# these should be escaped with a: \ or the name should be in double quotes e.g. "HARDWARE_DRIVER*.c"
sourceFiles:  
  - NUCLEO-WB55/**
  - Core/**
  - Middlewares/**
  - TouchGFX/**
  - Displays/**
  - Sensors/**

# When no makefile is present it will show a warning pop-up.
# However when compilation without the CubeMX Makefile is desired, this can be turned of.
suppressMakefileWarning: false
```

jortbmd, 2021-11-17:
Hi!
I am not all the familiar with TouchGFX, however it seems that the library file you have included is incorrectly named as it starts with a: ":". So double check the libraries you include. E.g. change the configuration below
```Yaml
# libraries to be included. The -l prefix to the library will be automatically added.
# Mind that non standard libraries should have a path to their respective directory.
libraries:
 - m
 - stdc++
 - supc++
 - :libtouchgfx-float-abi-hard.a
```
To the following configuration:
```Yaml
# libraries to be included. The -l prefix to the library will be automatically added.
# Mind that non standard libraries should have a path to their respective directory.
libraries:
 - m
 - stdc++
 - supc++
 - libtouchgfx-float-abi-hard
 ```

mikenorgate, 2021-11-19:
I resolved this by adding `- -fno-rtti ` to the `cxxFlags`

closed 2021-11-19 by mikenorgate