bmd-studio/stm32-for-vscode #88: Makefile generation and compilation flags issues

ANauzet, 2021-11-17:
Hello,
When generating for a C++ project, the CFLAGS are include in the C compilation:

`$(BUILD_DIR)/%.o: %.cpp Makefile | $(BUILD_DIR) 
	$(CXX) -c $(CXXFLAGS) $(CFLAGS) -Wa,-a,-ad,-alms=$(BUILD_DIR)/$(notdir $(<:.cpp=.lst)) $< -o $@`

So it is not possible to have C specific flags in CFLAGS because it will generate warning or error at CXX compilation. CFLAGS should be used only for CC, and CXXFLAGS should only be used for CXX

Same, ASFLAGS are not used in .s file compilation
`$(BUILD_DIR)/%.o: %.s Makefile | $(BUILD_DIR)
	$(AS) -c $(CFLAGS) $< -o $@`

jortbmd, 2021-11-17:
Hi!
Thanks for noticing will take a look at this and change this as you suggested.

jortbmd, 2022-03-08:
Hi. This should be fixed in the newest releases. Could you check out if this works in your situation?

jortbmd, 2022-06-29:
I am closing this, as this should have been fixed in the newer release. Should you continue to have issue feel free to reopen this or create a new issue.

closed 2022-06-29 by jortbmd