bmd-studio/stm32-for-vscode #89: C++ projects with a main.c

ANauzet, 2021-11-18:
When generating a project with TouchGFX for example, the project contain c++ sources, but the main is main.c

If selecting language as C in the Yaml config file, C++ files are not included in the Makefile
If selecting a C++ project, as there is no main.cpp, C project is automatically set.

Changing main.c to main.cpp is not an easy solution, as there are a lot of .c file in the project that will be change back to .c when generating with MX

It should be possible to have a C++ project but with a main.c file

jortbmd, 2021-11-18:
Hiya!

This has been a thing on my todo list for a while. So I have created something for the newest version. When you only have a main.c file in your project and the project settings set to .cpp it will generate the .cpp file for you in the build directory and compile with that. This way there is no need anymore to manually rename the main.c to main.cpp. I have created a pre-release for this function, you can download it here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.1.5-compilation-as-cpp
to install it press ctrl+shift+p or command+shift+p and select Extensions: Install from VSIX to install this file.

ANauzet, 2021-11-18:
Hello,
Thank you for the quick response. I just try the new version you provide.

It seems to work fine until the .elf generation. As there are cpp compiled file in the project, the elf generation should use $(CXX) instead of $(CC)

jortbmd, 2021-11-18:
Hi,

As far as I can tell it should switch from ${CC} to ${CXX} when generating the final elf file in the STM32Make.make file. Do note that you have to set the language in the STM32-for-VSCode.config.yaml file to: C++. 

At least I am guessing you are talking about this line: 
```Makefile
$(BUILD_DIR)/$(TARGET).elf: $(OBJECTS) STM32Make.make
	$(CXX) $(OBJECTS) $(LDFLAGS) -o $@
	$(SZ) $@
```

For the generation part, I do not see sufficient reason for implementation, as it would defeat the purpose of the extension auto generating. Having said that it is quite easy to do what you are suggesting. You can create your own makefile and call make -j 16 -f PATHTOYOURMAKEFILE in your terminal which should work just fine. Another option would be to create a task configuration so you can use the regular cmd/ctrl + shift + b to build it. An example of this can be found below  where you should replace "CustomMakefile.make" with your own makefile name:
```JSON
{
	"label": "Build with Custom Makefile",
	"type": "shell",
	"command": "make -j 16 -f CustomMakefile.make", 
	"options": {
		"cwd": "${workspaceRoot}"
	},
	"group": {
		"kind": "build",
		"isDefault": true
	},
	"problemMatcher": [
		"$gcc"
	]
},
```
And then for the debugging configuration in launch.json you can use the example below where you need to replace the executable name with the path to your .elf file.
```JSON
{
      "showDevDebugOutput": true,
      "cwd": "${workspaceRoot}",
      "executable": "./build/YOUR_TARGET_NAME.elf",
      "name": "Debug STM32",
      "request": "launch",
      "type": "cortex-debug",
      "servertype": "openocd",
      "preLaunchTask": "Build with Custom Makefile",
      "device": "stlink",
      "configFiles": [
        "openocd.cfg"
      ]
   }
```

jortbmd, 2022-01-13:
I am closing this due to it seeming that the issues is resolved. Should you experience any issues feel free to open up a new one ore comment on this one.

closed 2022-01-13 by jortbmd