bmd-studio/stm32-for-vscode #94: "device" value in launch.json is wrong (Can't see peripherals in debug)

RBahrami, 2022-02-16:
The value of "device" in launch.json is set to "stlink", it must be set to the microcontroller's name (without Tx suffix, like "STM32F429VI").

jortbmd, 2022-02-16:
Hi!
Thanks for opening an issue. You are right! I will look at this and include it in a next release.

RBahrami, 2022-03-10:
Hi,
There is still a small problem with the device name, for example, my chip is STM32F429VI, but in lunch.json, device will be set to STM32F429 (without the "VI" suffix).

closed 2022-06-29 by jortbmd

neosis, 2022-07-30:
@RBahrami / @jortbmd  what was the exact value that you provided to make it work?

I am using NUCLEO-F030R8 board with mounted device STM32F030R8T6

I am using following launch.json but still no luck.
`{
    "configurations": [
        {
            "showDevDebugOutput": "parsed",
            "cwd": "${workspaceRoot}",
            "executable": "./build/blink_led.elf",
            "name": "Debug STM32",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "preLaunchTask": "Build STM",
            "device": "STM32F030R8T6.s",
            "configFiles": [
                "openocd.cfg"
            ]
        },
        {
            "showDevDebugOutput": "parsed",
            "cwd": "${workspaceRoot}",
            "executable": "./build/blink_led.elf",
            "name": "Attach STM32",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "openocd",
            "preLaunchTask": "Build STM",
            "device": "STM32F030R8T6.s",
            "configFiles": [
                "openocd.cfg"
            ]
        }
    ]
}`

Any pointers on what am I missing?

Best
Ashutosh

neosis, 2022-07-30:
I downloaded the svd file for STM32F030 manually and kept in root of working folder. I also added "svdFile" property to both Attach and debug configuration objects in launch.json. Not sure if this is right way to do it - however seems to be working based on simple initial checks on peripheral port register values.