bmd-studio/stm32-for-vscode #96: C++ files not being debugged

AlexanderCtbk, 2022-03-03:
Hi

When building and debugging a fresh CubeMX project, everything works as expected including debugging.
However, when converting the main.c to main.cpp, I can't get breakpoints to work for cpp files, while existing .c files still allow breakpoints.
The debugger instead shows a disassembly of the cpp files. Setting a breakpoint in cpp source files shows "No Source file found projectpath/Core/Src/main.cpp

Would you have any pointers in how to fix this? I always try with a clean build and applied no code changes after the conversion from c to c++.

Thanks for your time!

jortbmd, 2022-03-08:
Hi!
Thank you for opening an issue. At the moment as I do not do a lot of c++ projects so I did not have the same issue or a direct solution. If I have time I will look at it and try to provide you with a solution. In the meanwhile could you perhaps provide me with your launch.json settings and the c_cpp_properties.json file, both located in the .vscode folder, as this might give a clue to what is wrong.

minigubben, 2022-03-08:
Hi,

I just dusted of an old hobby project today and had this problem aswell and after some digging it seems to be realated to the cflags used when building for debug.
In the generated STM32Make.make file there is this section:
```
ifeq ($(DEBUG), 1)
CFLAGS += -g -gdwarf-2
endif
```
Changing it to this seems to fix the issue:
```
ifeq ($(DEBUG), 1)
CFLAGS += -g -gdwarf-2
CXXFLAGS += -g -gdwarf-2
endif
```
However the makefile is generated from the plugin and it does not take this setting from the original makefile so i just added the line above to the tmplate inside the extension.js file in wait for a official patch.

I'm really not a C programmer so i don't know if this breaks anything else, but it seems to work so far.

Awesome extension bdw :)

jortbmd, 2022-06-30:
Hi! This has been fixed in the newest. Release. And therefore I am closing this issue. Should you have any further issues with this feel free to reopen this issue or make a new one.

closed 2022-06-30 by jortbmd