bmd-studio/stm32-for-vscode #97: There seems to be something wrong with the default FPU configuration

Simgor001, 2022-03-08:
Hi, I'm a beginner and I love this plugin. But there is one problem that has been bothering me. The MCU I use is STM32F103VET6, and when I build it will report an error that looks like this:
[image]

Here is part of my Makefile:

```makefile
Here is part of my Makefile:
#######################################
# CFLAGS
#######################################
# cpu
CPU = -mcpu=cortex-m3

# fpu
# NONE for Cortex-M0/M0+/M3

# float-abi

# mcu
MCU = $(CPU) -mthumb $(FPU) $(FLOAT-ABI)
```

It is written here that Cortex-M0/M0+/M3 has no FPU, so the configuration of the fpu in the "STM32-for-VSCode.config.yaml" file looks like this:

```yaml
# MCU settings

targetMCU: stm32f1x

cpu: cortex-m3 # type of cpu e.g. cortex-m4

fpu: soft # Defines how floating points are defined. Can be left empty.

floatAbi:

ldscript: STM32F103VETx_FLASH.ld # linker script
```
If I manually delete the FPU configuration in the "STM32-for-VSCode.config.yaml" file or configure it to "fpu: auto", the project can be compiled normally because "arm-none-eabi-gcc" The command prompt does not have the "-mfpu=soft" configuration, but has the "-mfpu=auto" configuration.

While it is possible to solve this problem manually, it would be even better if it could be done automatically! Thanks!

jortbmd, 2022-03-08:
Hi!
Thank you for opening up this issue. You are right. I have implemented a fix for this and released a new version with this fix under version 3.2.1. If you could check it out it would be greatly appreciated.

Simgor001, 2022-03-09:
Thanks, it's working fine now.

closed 2022-03-09 by Simgor001