bmd-studio/stm32-for-vscode #98: Linker flag behavior prevents compilation

kutukvpavel, 2022-03-13:
The extension suddenly started to put nano.specs into the generated makefile, instead of nosys.specs, leading to infamous linker errors like `undefined reference to _exit` (and `_sbrc` and other POSIX stuff).

I don't really understand what changed, except I installed the february VSCode update yesterday.

What I found out during investigation is: STM32-for-VSCode.config.yml did not change, `assemblyFlags:  - -specs=nosys.specs` was and is there. CubeMX-generated Makefile did not change either, and it in fact contains `LDFLAGS = -specs=nano.specs`. Sure enough, when I edited the CubeMX-generated Makefile to `nosys`, the problem disappeared. However, the Makefile is reverted back to `nano` every time I regenerate the code using CubeMX.

So, it seems for some reason extension configuration started to fail to override original Makefile linker options. Or am I missing something?

Germwalker, 2022-03-26:
I agree, there is a problem with the way this flag is added to the makefile.
I solved the problem this way :

# Additional LD Flags from config file
ADDITIONALLDFLAGS = -specs=nano.specs -specs=nosys.specs

jortbmd, 2022-03-28:
Hi!
This can happen. In the past the nosys.specs was always added by default. However this affected code size. So I decided to not add this as a default anymore.
The simple fix is to add -specs=nosys.specs to the linkerFlags option in the STM32-for-VSCode.config.yaml. E.g.:
```yaml
# Compiler flags
cFlags: []
cxxFlags: []
assemblyFlags: []
linkerFlags:
  - -specs=nano.specs
```

closed 2022-06-29 by jortbmd