bmd-studio/stm32-for-vscode #101: Option to add flags to make command

infinity1096, 2022-03-31:
When executing the current makefile, each gcc/g++ command is printed out, flooding the compile errors and warnings in between. Could there be an option to invoke make with the silent flag ("make ... -s") to only show the warning? 

Thank you for developing this great plugin.

jortbmd, 2022-04-06:
Hi!
That might be a nice idea. I want to overhaul the build system in the future a bit and will think about this.

jortbmd, 2022-11-14:
I will add this feature in the next release, which will probably be released this week. A preview can be found here:  https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.2-%23101-%23116-%23119 and you can use the attached .vsix file to install it by pressing CMD/CTRL+shift+P and by typing and selecting Extensions: Install from VSIX.

Then add:

makeFlags:
 - -O
to your STM32-for-VSCode.config.yaml

jortbmd, 2022-12-08:
The new release is out and this feature is added. Hope this works for you. If you find any issues feel free to open up a new issue or re-open this one.

closed 2022-12-08 by jortbmd