bmd-studio/stm32-for-vscode #102: Import CubeIDE project "No source files" error

PNPaCo, 2022-04-11:
Hey mate,

I've been handed a project which was made in the [STM Cube IDE](https://www.st.com/en/development-tools/stm32cubeide.html), and I would like to continue working on this project in VSCode.

I believe this is the option for me:
[image]

But then I'm presented with this error:
[image]

I seem to have 3 "project" files, and they all seem to have a lot of things inside that could be considered sources - so I'm not sure what the problem is
[image]

Looking around online hasn't returned me any promising leads so I'm hoping you can help get me going.

Cheers!

jortbmd, 2022-04-11:
Hi!
In the formats of the projects I have tested the .project file was populated with sources. However I did some more testing with fresh projects and this does not seem to be the case anymore. I will look into this if there is a way that the migration can be done automatically, without any additional configuration.
For now if it is a STM32CubeMX project and contains and .ioc file. Switch the Toolchain/IDE to Makefile and regenerate. This can be done in the project manager tab.
If this is not possible you can  manually add all the necessary files and configuration options to the STM32-for-VSCode.config.yaml file, which might be a bit tedious. The correct configuration can be copied from the STM32CubeIDE build configuration. A list  of sources can be found in the Debug or Release folder of the STM32CubeIDE project after building it. They have .c files represented in a list. These can be copied to the STM32-for-VSCode.config.yaml file. These files are named subdir.mk.

Hope this helps.

PNPaCo, 2022-04-12:
Ok, for future people who might encounter this issue, here's my solution:

1. Download [CubeMX](https://www.st.com/en/development-tools/stm32cubemx.html#st-get-software)
2. Open the .ioc file in the original STMCubeIDE folder
3. Go to project manager tab
4. Change to makefile
5. Generate code
6. Set the programmer for STM32 for VSCode
7. Profit!

The missing step is that the CubeMX program is _inside_ CubeIDE, but you can't change it to generate a makefile. You have to download the standalone CubeMX program for that to work

jortbmd, 2022-04-12:
Hi! Thank for clarifying my earlier post. It is greatly appreciated. I will leave this issue open for now, until I have implemented a way to fix the importing.

yamaan93, 2023-04-26:
I just wanted to chime in since I started randomly experiencing this problem as well. I did notice however that if you just click build, it seems to build correctly on it's own without importing the project. Not sure how or why, but it works.