bmd-studio/stm32-for-vscode #103: OpenOCD GDB Server Quit Unexpectedly While Trying to Debug

dbacher42, 2022-04-22:
I'm trying to set up STM32 for VSCode to program a Nucleo F401RE board. I can build and flash just fine, but when I try to open debug mode I get an error saying that the OpenOCD GDB Server quit unexpectedly.

Here is the output of the terminal:
[image]

Here is the output of the debug terminal:
[image]

Here is the launch.json file:
[image]

jortbmd, 2022-04-22:
Hi!
thank you for opening an issue. Given the output on the debug terminal and regular terminal I can't really tell what causes the crash. The only discrepancy I can see is in the configuration. Could you perhaps delete the line with: ``"device": "stm32f401xe.s"`` and check again. This might fix the problem.
Also just to be sure try to update you installation of openocd. The extension can do this for you by using the command: "STM32: install all the build tools for the STM32 for VSCode extension". Commands can be run using the keyboard combination: CTRL/CMD + shift + p
If not; change the showDevDebugOutput line too  ``"showDevDebugOutput": "raw"``. This way it will show the raw output of GDB which might contain more information on the particular error, which might me help further debug this.

dbacher42, 2022-04-22:
Thanks for your response,

I tried deleting that line in the configuration file and that did not work. I already have openocd installed, but just in case I ran the command to install build tools and got the following error when trying to install openocd:
[image]

I also changed the showDevDebugOutput line, but it looks like the output to the debug terminal is the same as it was before:
[image]

jortbmd, 2022-04-29:
Thanks  for your response. I can't really tell what is going wrong with opening openocd. As the terminal gives very little information about why it closes. Usually it at least tries to create a connection to the board and shows a failure when it does not succeed. 
The installation failure on windows is a different issue, however I will look into this, as this as worked inconsistently for a couple of people on Windows.
I can see you are trying to debug an RTOS. This might be causing an issue as this is the only real difference between flashing and debugging (flashing does not care that it is an RTOS). You could try settings the -rtos auto flag in the debuggerArgs of thew  launch file. e.g.:
```
{
  "showDevDebugOutput": "parsed",
  "cwd": "${workspaceRoot}",
  "executable": "./build/your_elf_file.elf",
  "name": "Debug STM32",
  "request": "launch",
  "type": "cortex-debug",
  "servertype": "openocd",
  "preLaunchTask": "Build STM",
  "configFiles": [
    "openocd.cfg",
  ],
  "debuggerArgs": [
    "-rtos auto"
  ]
},
```

jortbmd, 2022-06-29:
I will close this because of inactivity. Should you still have issues do not hesitate to reopen this issue.

closed 2022-06-29 by jortbmd