bmd-studio/stm32-for-vscode #105: STM32G0 vanilla build fails because of stm32g0xx.h

Germwalker, 2022-04-30:
Hello, 

I tried to build a firmware generated with CubeMX for the STM32f070RBT6 MCU.
The build fails because of a missing file : **stm32g0xx.h**. 

The file has been generated and is located inside **Drivers/CMSIS/Device/ST/STM32G0xx/Include/stm32g0xx.h**
The makefile contains the path as **-IDrivers/CMSIS/Device/ST/STM32G0xx/Include**

I don't understand why the build fails.

Here is the console trace : 

```
".config/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/10.3.1-2.3.1/.content/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m0plus -mthumb   -DCMSIS_device_header=<stm32g0xx.h> -DSTM32G070xx -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32G0xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32G0xx_HAL_Driver/Inc -IDrivers/STM32G0xx_HAL_Driver/Inc/Legacy -IMiddlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2 -IMiddlewares/Third_Party/FreeRTOS/Source/include -IMiddlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM0 -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/main.d" -Wa,-a,-ad,-alms=build/main.lst Core/Src/main.c -o build/main.o
/bin/sh: ligne 1: stm32g0xx.h: Aucun fichier ou dossier de ce type
/bin/sh: ligne 1: stm32g0xx.h: Aucun fichier ou dossier de ce type
/bin/sh: ligne 1: stm32g0xx.h: Aucun fichier ou dossier de ce type
make: *** [STM32Make.make:229 : build/adc.o] Erreur 1
make: *** Attente des tâches non terminées....
make: *** [STM32Make.make:229 : build/app_freertos.o] Erreur 1
make: *** [STM32Make.make:229 : build/dma.o] Erreur 1
/bin/sh: ligne 1: stm32g0xx.h: Aucun fichier ou dossier de ce type
/bin/sh: ligne 1: stm32g0xx.h: Aucun fichier ou dossier de ce type
make: *** [STM32Make.make:229 : build/i2c.o] Erreur 1
make: *** [STM32Make.make:229 : build/main.o] Erreur 1
make: *** [STM32Make.make:229 : build/gpio.o] Erreur 1
Échec du lancement du processus de terminal "/usr/bin/bash '-c', '"/usr/bin/make" -j16 -f STM32Make.make '" (code de sortie : 2).
```

RootShell-coder, 2022-05-03:
_This REFLOW-HEATER.zip archive is empty. It contains REFLOW-HEATER.ioc size 4KB created in CubeMX v6.3.0
There really is no such file there stm32**g**0xx.h. Cubemx creates stm32**f**0xx.h. I can't say whether the version of this file with g was._

jortbmd, 2022-05-03:
Hi! Thanks for reaching out!
I get the same error. I tried to compile it and failed. Even with the ST provided makefile I get the same error. Everything seems to be in order however when using CMSIS_V1 it seems to work. The definition of the device header is the issue:``- CMSIS_device_header=<stm32g0xx.h>``. When this is converted to: ``- CMSIS_device_header=\"stm32g0xx.h\"`` the error is fixed. You can change this in the Makefile and STM32 for VSCode will use this definition. E.g.:
```makefile
# C defines
C_DEFS =  \
-DUSE_HAL_DRIVER \
-DSTM32G070xx \
-DCMSIS_device_header=\"stm32g0xx.h\"
```
This seems to be an issue with how CubeMX generates the code. As far as I know the CMSIS_V2 support is still experimental. I would advise you and be grateful to you if you could make a bug report for ST as this would help the community as a whole. This can be done here: https://community.st.com/s/topic/0TO0X000000BWVLWA4/bugreport